llvm/circt #2479: Integration test broken

youngar, 2022-01-19:
The nightly integration tests are failing to configure.  I think this is probably a result of upgrading the test image here https://github.com/llvm/circt/commit/d2ff58a59cdcb0e925c47fe76c6afe3505e7c034.

```
CMake Error at cmake/modules/HandleLLVMOptions.cmake:295 (message):
  Host compiler does not support '-fuse-ld=lld'
Call Stack (most recent call first):
  CMakeLists.txt:737 (include)

-- Configuring incomplete, errors occurred!
```

teqdruid, 2022-01-19:
We also upgraded from v6 to v10, which is quite a jump.

jopperm, 2022-01-19:
The biggest [change](https://github.com/circt/images/compare/v6-beta3...v10) to the integration image seems to be manual installation of LLVM, but I'll need some time to dig into that. I'll revert back to the old image for now.

jopperm, 2022-01-30:
I think I found the issue, see the PR above. Once that lands, I'll introduce a new integration-image-bump-PR.

closed 2022-01-30 by teqdruid