llvm/circt #2491: [SCFToCalyx] Crash with external memories

mikeurbach, 2022-01-21:
I'm trying to run the following program through SCFToCalyx:

```mlir
 func @main(%arg0 : memref<64xi32>, %arg1 : memref<64xi32>) -> i32 {
    %c0_i32 = arith.constant 0 : i32
    %c0 = arith.constant 0 : index
    %c64 = arith.constant 64 : index
    %c1 = arith.constant 1 : index
    %0:2 = scf.while (%arg2 = %c0, %arg3 = %c0_i32) : (index, i32) -> (index, i32) {
      %1 = arith.cmpi slt, %arg2, %c64 : index
      scf.condition(%1) %arg2, %arg3 : index, i32
    } do {
    ^bb0(%arg2: index, %arg3: i32):  // no predecessors
      %1 = arith.addi %arg2, %c1 : index
      %2 = memref.load %arg0[%arg2] : memref<64xi32>
      %3 = memref.load %arg1[%arg2] : memref<64xi32>
      %4 = arith.muli %2, %3 : i32
      %5 = arith.addi %arg3, %4 : i32
      scf.yield %1, %5 : index, i32
    }
    return %0#1 : i32
  }
```

Unfortunately, this seems to crash the pass, and worse, it silently just exits with a return code of 1. This appears to be related to the external memories; if I move the parameters to internal allocations, the pass succeeds:

```mlir
  func @main() -> i32 {
    %arg0 = memref.alloc() : memref<64xi32>
    %arg1 = memref.alloc() : memref<64xi32>
```

I haven't begun looking into this yet, but will mention here if I do.

mortbopet, 2022-01-22:
strange; dumping the IR after the SCFToCalyx pass looks fine (there's a lot, I cannot rule out something fishy in there). Stepping through that point, it seems like the verifier fails for some reason. This is also the case if saving the IR dump and passing it to `circt-opt` - failing silently with no info as to why.

mikeurbach, 2022-01-23:
Interesting, sounds like we should be looking for places the verifier can silently fail? I can try debugging this.

mikeurbach, 2022-01-24:
@mortbopet I think I've gotten to the bottom of it. For one, the Calyx slice and pad verifiers didn't emit any error message when they failed, so I fixed that. This pointed out an issue in SCFToCalyx where we unconditionally sliced a signal, but might also need to pad it. After fixing that I found SCFToCalyx was using one bit per element for external memref addresses, when the normal thing (which we do for internal memrefs) is to use log2 of the number of elements. I pushed these changes in the following patches, please take a look: https://github.com/llvm/circt/compare/15fd490735e987287f98c69b41ad048b90405864...05c1a85333f77cd959bc5874a58b7daeca194c08

mortbopet, 2022-01-24:
Great find! commits looks good to me :+1: .

closed 2022-01-24 by mikeurbach