llvm/circt #2514: [FIRRTL] (core dump) leq,SInt<0>

drom, 2022-01-24:
Running following fir file

```python
; seed: 3339
circuit top_mod :
  module top_mod :
    input clock1: Clock
    input inp_a: {inp_c: SInt<0>, inp_f: {inp_g: {inp_j: SInt<0>}}, inp_h: {inp_k: UInt<15>}}
    reg tmp23: UInt, clock1
    tmp23 <= head(UInt<1>("b1"), 0)
    node _tmp26 = leq(inp_a.inp_f.inp_g.inp_j, asSInt(tmp23))
```

with `firtool a_top_mod.fir --verilog -o=a_top_mod.fir.mfc.v`

I am getting this error:
```
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/APInt.cpp:284: int llvm::APInt::compareSigned(const llvm::APInt&) const: Assertion `BitWidth == RHS.BitWidth && "Bit widths must be same for comparison"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --verilog --lowering-options=disallowPackedArrays,noAlwaysComb,disallowLocalVariables -o=a_top_mod.fir.mfc.v
 #0 0x000000000045c441 PrintStackTraceSignalHandler(void*) Signals.cpp:0:0
 #1 0x0000000000459d0e SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f3fc26d1420 __restore_rt (/lib64/libc.so.6+0x56420)
 #3 0x00007f3fc272124c __pthread_kill_implementation (/lib64/libc.so.6+0xa624c)
 #4 0x00007f3fc26d1356 gsignal (/lib64/libc.so.6+0x56356)
 #5 0x00007f3fc26ba7d3 abort (/lib64/libc.so.6+0x3f7d3)
 #6 0x00007f3fc26ba6eb _nl_load_domain.cold (/lib64/libc.so.6+0x3f6eb)
 #7 0x00007f3fc26c9946 (/lib64/libc.so.6+0x4e946)
 #8 0x000000000046733f llvm::APInt::compareSigned(llvm::APInt const&) const (../../llvm/circt/build/bin/firtool+0x46733f)
 #9 0x000000000075dc59 circt::firrtl::LEQPrimOp::fold(llvm::ArrayRef<mlir::Attribute>) (../../llvm/circt/build/bin/firtool+0x75dc59)
#10 0x000000000073d2d3 mlir::LogicalResult llvm::detail::UniqueFunctionBase<mlir::LogicalResult, mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&>::CallImpl<std::enable_if<(llvm::disjunction<std::is_same<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp>, mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::OpTrait::ZeroSuccessor<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::OpTrait::NOperands<2u>::Impl<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, circt::firrtl::SameOperandsIntTypeKind<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::InferTypeOpInterface::Trait<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > >, circt::firrtl::LEQPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >, mlir::MemoryEffectOpInterface::Trait<mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> > > >::value) && (llvm::detail::detector<void, mlir::Op<circt::firrtl::LEQPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::NOperands<2u>::Impl, circt::firrtl::SameOperandsIntTypeKind, mlir::InferTypeOpInterface::Trait, mlir::MemoryEffectOpInterface::Trait>::has_single_result_fold, mlir::OpTrait::OneResult<circt::firrtl::LEQPrimOp> >::value_t::value), llvm::unique_function<mlir::LogicalResult (mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const> >::type mlir::Op<circt::firrtl::LEQPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::NOperands<2u>::Impl, circt::firrtl::SameOperandsIntTypeKind, mlir::InferTypeOpInterface::Trait, mlir::MemoryEffectOpInterface::Trait>::getFoldHookFnImpl<circt::firrtl::LEQPrimOp>()::'lambda'(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const>(void*, mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) (../../llvm/circt/build/bin/firtool+0x73d2d3)
#11 0x00000000006ec4ba mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) (../../llvm/circt/build/bin/firtool+0x6ec4ba)
#12 0x000000000086a519 mlir::OperationFolder::tryToFold(mlir::OpBuilder&, mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::function_ref<void (mlir::Operation*)>) (../../llvm/circt/build/bin/firtool+0x86a519)
#13 0x000000000086b8a3 mlir::OperationFolder::tryToFold(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, llvm::function_ref<void (mlir::Operation*)>, bool*) (../../llvm/circt/build/bin/firtool+0x86b8a3)
#14 0x00000000008562b9 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>) (.constprop.0) GreedyPatternRewriteDriver.cpp:0:0
#15 0x000000000085842a mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) (../../llvm/circt/build/bin/firtool+0x85842a)
#16 0x0000000000908846 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x908846)
#17 0x0000000000908dd8 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x908dd8)
#18 0x00000000009072ce mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (../../llvm/circt/build/bin/firtool+0x9072ce)
#19 0x00000000009088cf mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x9088cf)
#20 0x0000000000908dd8 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x908dd8)
#21 0x00000000009072ce mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (../../llvm/circt/build/bin/firtool+0x9072ce)
#22 0x00000000009088cf mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x9088cf)
#23 0x0000000000908dd8 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x908dd8)
#24 0x000000000090a42b mlir::PassManager::run(mlir::Operation*) (../../llvm/circt/build/bin/firtool+0x90a42b)
#25 0x000000000040d8e1 processBuffer(mlir::MLIRContext&, mlir::TimingScope&, llvm::SourceMgr&, llvm::Optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile> > >&) firtool.cpp:0:0
#26 0x000000000040e874 processInputSplit(mlir::MLIRContext&, mlir::TimingScope&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::Optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile> > >&) firtool.cpp:0:0
#27 0x0000000000409aa1 main (../../llvm/circt/build/bin/firtool+0x409aa1)
#28 0x00007f3fc26bb5c0 __libc_start_call_main (/lib64/libc.so.6+0x405c0)
#29 0x00007f3fc26bb67c __libc_start_main@GLIBC_2.2.5 (/lib64/libc.so.6+0x4067c)
#30 0x0000000000409e25 _start /home/abuild/rpmbuild/BUILD/glibc-2.34/csu/../sysdeps/x86_64/start.S:118:0
```

youngar, 2022-01-25:
Golfed down a tiny bit more:
```
circuit top_mod :
  module top_mod :
    input a: UInt<0>
    output c: UInt<1>

    wire w: UInt<0>
    w <= UInt<1>(1)
    c <= leq(a, w)
```

seldridge, 2022-01-25:
This is trying to do a comparison of an unsigned greater than or equal (`APInt(0, 0) > (APInt(1, 0)`) and it's dying because the widths have to match. The issue is happening here: https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/FIRRTLFolds.cpp#L572

Note: the original example is failing on line 565 (the signed variant of the same case). The unsigned variant is probably a small change to the `getMaxUnsignedValue` utility to return a zero-width APInt. However, the signed variant is going to need special casing or special utilities as APInts will error out all sorts of ways when working with signed numbers that hit zero width cases as LLVM utilities will try to sign extend these.

dtzSiFive, 2022-05-03:
This still fails, and FWIW here are updated (and pinned, fun tip: press 'y' when viewing code and GitHub pins to current revision for future-proofing/permalinking) locations where the two examples fail:

https://github.com/llvm/circt/blob/be3f5b530bb69dc64656b662bded7e65a73d55e4/lib/Dialect/FIRRTL/FIRRTLFolds.cpp#L547
https://github.com/llvm/circt/blob/be3f5b530bb69dc64656b662bded7e65a73d55e4/lib/Dialect/FIRRTL/FIRRTLFolds.cpp#L554

PR #3204 (2022-05-27): [FIRRTL] Fix folds of zero-width values with zero-width constants
Fixes #2514.

In general, these range checks will work at larger-than-needed `commonWidth`, FWIW, but that's unnecessary/overkill for normal case.  Address the zero-width on both sides issue (`commonWidth==0`) by ensuring we compare at 1-bit or wider since various AP(S)Int comparisons don't work nicely with zero-bits (and our getMin/getMax return 1-bit values).

(This also removes one need for `orSelf` variants of the extension, as we'd be requesting an extension of the min/max 1-bit values to the common zero width.)

Add exhaus [... cut]
commit cbee3460 (2022-05-27): [FIRRTL] Fix folds of zero-width values with zero-width constants (#3204)

fix folds of zero-width and zero-width const

Fix by ensuring the commonWidth used for checks
is always at least 1.

Fixes #2514.
closed 2022-05-27 by dtzSiFive