llvm/circt #2529: [SV] Verification failures for inout operations with type alias

uenoku, 2022-01-27:
```mlir
  hw.type_scope @_Foo  {
    hw.typedecl @_Foo_type0 : !hw.struct<io: i1>
  }
  hw.module @Foo() -> () {
    %.b.output = sv.wire  : !hw.inout<typealias<@_Foo_type0, !hw.struct<io: i1>>>
    %0 = sv.struct_field_inout %.b.output["io"] : !hw.inout<typealias<@_Foo_type0, !hw.struct<io: i1>>>
    hw.output
  }
```

Currently this raises a verification failure. 
```
bar.mlir:7:10: error: 'sv.struct_field_inout' op operand #0 must be an inout type with struct field, but got '!hw.inout<typealias<@_Foo_type0, !hw.struct<io: i1>>>'
    %0 = sv.struct_field_inout %.b.output["io"] : !hw.inout<typealias<@_Foo_type0, !hw.struct<io: i1>>>
```
This is caused by the ods definition of `InOutStructType`. It is necessary to look at an inner types or a canonical type of type alias. 

This problem is also applicable to other inout operations like array_index_inout, part_select_inout.

mikeurbach, 2022-01-27:
I haven't been keeping up with these additions, but it should be sufficient to use `hw::type_isa` and `TypeAliasOr` in the ODS predicate definitions, like was done for the original definition of StructType: https://github.com/llvm/circt/blob/fe2fdb7475b5dc9948bf17a2513a51844fb40da3/include/circt/Dialect/HW/HWTypes.td#L47

uenoku, 2022-01-27:
One different point is that `InOutStructType` is trying to match a type `hw.inout<hw.struct<...>>`. 
Does this work `TypeAliasOr` directly? I'm wondering I have to create `TypeAliasInoutOr` so that it recognizes the nested type.

mikeurbach, 2022-01-28:
I just took a quick glance, would it be possible for the InOutStructType predicate to be the following? I can take a look a bit later.

```
hw::type_isa<hw::StructType>(getInOutElementType($_self))"
```

uenoku, 2022-01-28:
It worked! Thanks a lot @mikeurbach!

closed 2022-01-28 by uenoku