llvm/circt #2535: How to pass attribute down to HW?

Kuree, 2022-01-28:
Hello,

I'm trying to add source-level debugging support to circt and I'm having problems to passing attribute information throughout the lowering process. Here is what I did:
1. Adding attribute to the AssignOp when Firrtl is lowered to HW:
https://github.com/Kuree/circt/blob/b7a9e4d90850e3953f99ef5347b79a1e675f865f/lib/Conversion/FIRRTLToHW/LowerToHW.cpp#L2269-L2276
2. Collect the attribute when HW is converted to SV:
https://github.com/Kuree/circt/blob/b7a9e4d90850e3953f99ef5347b79a1e675f865f/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L2599-L2602

Unfortunately this didn’t work: none of the attributes I added earlier appear in the conversion pass. What is the canonical way to pass attribute information from one dialect to another?

Thanks,
Keyi

youngar, 2022-01-28:
You might try running `firtool --print-ir-after-all` and try to spot what pass is deleting the debug attribute. In particular, the node op in your test code is probably not being assigned a symbol, so it is probably being deleted.

The following example might be a bit more promising:
```
circuit Test: %[[{"class":"firrtl.transforms.DontTouchAnnotation", "target":"~Test|Test>z"}]]
  module Test:
    input x: {a: UInt<2>}
    output y: UInt<2>
    node z = x
    y <= z.a
 ```
 after parsing:
 ```
 firrtl.module @Test(in %x: !firrtl.bundle<a: uint<2>>, out %y: !firrtl.uint<2>) {
  %z = firrtl.node sym @z %x  : !firrtl.bundle<a: uint<2>>
  %0 = firrtl.subfield %z(0) : (!firrtl.bundle<a: uint<2>>) -> !firrtl.uint<2>
  firrtl.connect %y, %0 : !firrtl.uint<2>, !firrtl.uint<2>
}
```
```
// -----// IR Dump After LowerFIRRTLTypes //----- //
firrtl.circuit "Test"   {
  firrtl.module @Test(in %x_a: !firrtl.uint<2> [{hw.debug.name = "x.a"}], out %y: !firrtl.uint<2>) {
    %z_a = firrtl.node sym @z_z_a %x_a  {hw.debug.name = "z"} : !firrtl.uint<2>
    firrtl.connect %y, %z_a : !firrtl.uint<2>, !firrtl.uint<2>
  }
}
```

 ```
// -----// IR Dump After LowerFIRRTLToHW //----- //
module  {
  hw.module @Test(%x_a: i2) -> (y: i2) {
    %z_a = sv.wire sym @z_z_a  : !hw.inout<i2>
    sv.assign %z_a, %x_a {hw.debug.name = "z"} : i2
    hw.output %x_a : i2
  }
}
```

Kuree, 2022-01-28:
Thanks. That's very helpful. the attributes disappeared after `LowerFIRRTLToHW`. It seems like I was pass attributes to the wrong places. I will let you know if I have more questions.

youngar, 2022-01-28:
@Kuree Any chance you could post your input program? It looks like the `NodeOp` lowering is working at least.

Kuree, 2022-01-28:
Sure thing:

```
circuit SimpleVendingMachineTester : 
  module ImplicitStateVendingMachine : 
    input clock : Clock
    input reset : Reset
    output io : {flip nickel : UInt<1>, flip dime : UInt<1>, dispense : UInt<1>}
    
    node _T = and(io.nickel, io.dime) @[SimpleVendingMachine.scala 19:22]
    node _T_1 = eq(_T, UInt<1>("h00")) @[SimpleVendingMachine.scala 19:10]
    node _T_2 = asUInt(reset) @[SimpleVendingMachine.scala 19:9]
    node _T_3 = or(_T_1, _T_2) @[SimpleVendingMachine.scala 19:9]
    node _T_4 = eq(_T_3, UInt<1>("h00")) @[SimpleVendingMachine.scala 19:9]
    when _T_4 : @[SimpleVendingMachine.scala 19:9]
      printf(clock, UInt<1>(1), "Assertion failed: Only one of nickel or dime can be input at a time!\n    at SimpleVendingMachine.scala:19 assert(!(io.nickel && io.dime), \"Only one of nickel or dime can be input at a time!\")\n") @[SimpleVendingMachine.scala 19:9]
      stop(clock, UInt<1>(1), 1) @[SimpleVendingMachine.scala 19:9]
      skip @[SimpleVendingMachine.scala 19:9]
    reg value : UInt<3>, clock with : (reset => (reset, UInt<3>("h00"))) @[ImplicitStateVendingMachine.scala 11:22]
    wire incValue : UInt<3>
    incValue <= UInt<3>("h00")
    node doDispense = geq(value, UInt<3>("h04")) @[ImplicitStateVendingMachine.scala 13:26]
    when doDispense : @[ImplicitStateVendingMachine.scala 15:21]
      value <= UInt<1>("h00") @[ImplicitStateVendingMachine.scala 16:11]
      skip @[ImplicitStateVendingMachine.scala 15:21]
    else : @[ImplicitStateVendingMachine.scala 17:16]
      node _value_T = add(value, incValue) @[ImplicitStateVendingMachine.scala 18:20]
      node _value_T_1 = tail(_value_T, 1) @[ImplicitStateVendingMachine.scala 18:20]
      value <= _value_T_1 @[ImplicitStateVendingMachine.scala 18:11]
      skip @[ImplicitStateVendingMachine.scala 17:16]
    when io.nickel : @[ImplicitStateVendingMachine.scala 21:20]
      incValue <= UInt<1>("h01") @[ImplicitStateVendingMachine.scala 21:31]
      skip @[ImplicitStateVendingMachine.scala 21:20]
    when io.dime : @[ImplicitStateVendingMachine.scala 22:18]
      incValue <= UInt<2>("h02") @[ImplicitStateVendingMachine.scala 22:29]
      skip @[ImplicitStateVendingMachine.scala 22:18]
    io.dispense <= doDispense @[ImplicitStateVendingMachine.scala 24:15]
    
  module SimpleVendingMachineTester : 
    input clock : Clock
    input reset : UInt<1>
    output io : {}
    
    inst dut of ImplicitStateVendingMachine @[SimpleVendingMachine.scala 73:19]
    dut.clock <= clock
    dut.reset <= reset
    reg cycle : UInt<4>, clock with : (reset => (reset, UInt<4>("h00"))) @[Counter.scala 60:40]
    wire done : UInt<1>
    done <= UInt<1>("h00")
    when UInt<1>("h01") : @[Counter.scala 118:17]
      node wrap_wrap = eq(cycle, UInt<4>("h09")) @[Counter.scala 72:24]
      node _wrap_value_T = add(cycle, UInt<1>("h01")) @[Counter.scala 76:24]
      node _wrap_value_T_1 = tail(_wrap_value_T, 1) @[Counter.scala 76:24]
      cycle <= _wrap_value_T_1 @[Counter.scala 76:15]
      when wrap_wrap : @[Counter.scala 86:20]
        cycle <= UInt<1>("h00") @[Counter.scala 86:28]
        skip @[Counter.scala 86:20]
      done <= wrap_wrap @[Counter.scala 118:24]
      skip @[Counter.scala 118:17]
    when done : @[SimpleVendingMachine.scala 76:15]
      node _T = bits(reset, 0, 0) @[SimpleVendingMachine.scala 76:21]
      node _T_1 = eq(_T, UInt<1>("h00")) @[SimpleVendingMachine.scala 76:21]
      when _T_1 : @[SimpleVendingMachine.scala 76:21]
        stop(clock, UInt<1>(1), 0) @[SimpleVendingMachine.scala 76:21]
        skip @[SimpleVendingMachine.scala 76:21]
      node _T_2 = bits(reset, 0, 0) @[SimpleVendingMachine.scala 76:29]
      node _T_3 = eq(_T_2, UInt<1>("h00")) @[SimpleVendingMachine.scala 76:29]
      when _T_3 : @[SimpleVendingMachine.scala 76:29]
        stop(clock, UInt<1>(1), 0) @[SimpleVendingMachine.scala 76:29]
        skip @[SimpleVendingMachine.scala 76:29]
[... 21 lines cut ...]
    dimeInputs[8] <= UInt<1>("h00") @[SimpleVendingMachine.scala 79:29]
    dimeInputs[9] <= UInt<1>("h01") @[SimpleVendingMachine.scala 79:29]
    wire expected : UInt<1>[10] @[SimpleVendingMachine.scala 80:29]
    expected[0] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[1] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[2] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[3] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[4] <= UInt<1>("h01") @[SimpleVendingMachine.scala 80:29]
    expected[5] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[6] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[7] <= UInt<1>("h01") @[SimpleVendingMachine.scala 80:29]
    expected[8] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    expected[9] <= UInt<1>("h00") @[SimpleVendingMachine.scala 80:29]
    dut.io.nickel <= nickelInputs[cycle] @[SimpleVendingMachine.scala 82:17]
    dut.io.dime <= dimeInputs[cycle] @[SimpleVendingMachine.scala 83:15]
    node _T_4 = eq(dut.io.dispense, expected[cycle]) @[SimpleVendingMachine.scala 84:26]
    node _T_5 = bits(reset, 0, 0) @[SimpleVendingMachine.scala 84:9]
    node _T_6 = or(_T_4, _T_5) @[SimpleVendingMachine.scala 84:9]
    node _T_7 = eq(_T_6, UInt<1>("h00")) @[SimpleVendingMachine.scala 84:9]
    when _T_7 : @[SimpleVendingMachine.scala 84:9]
      printf(clock, UInt<1>(1), "Assertion failed\n    at SimpleVendingMachine.scala:84 assert(dut.io.dispense === expected(cycle))\n") @[SimpleVendingMachine.scala 84:9]
      stop(clock, UInt<1>(1), 1) @[SimpleVendingMachine.scala 84:9]
      skip @[SimpleVendingMachine.scala 84:9]

```

youngar, 2022-01-28:
Since you only add debug information to aggregate types in `firrtl-lower-types`, it looks like all the limited amount of debug information being attached is just being optimized way in the example.

By the time we get to LowerToHW there are only two pieces of debug information still attached to the IR:
```
    %dut_clock, %dut_reset, %dut_io_nickel, %dut_io_dime, %dut_io_dispense = firrtl.instance dut  @ImplicitStateVendingMachine(in clock: !firrtl.clock, in reset: !firrtl.uint<1>, in io_nickel: !firrtl.uint<1>, in io_dime: !firrtl.uint<1>, out io_dispense: !firrtl.uint<1>)
    firrtl.connect %dut_clock, %clock {hw.debug.name = "dut"} : !firrtl.clock, !firrtl.clock
    firrtl.connect %dut_reset, %reset {hw.debug.name = "dut"} : !firrtl.uint<1>, !firrtl.uint<1>
 ```
 
 After LowerToHW this becomes:
```mlir
  %dut.io_dispense = hw.instance "dut" @ImplicitStateVendingMachine(clock: %clock: i1, reset: %reset: i1, io_nickel: %12: i1, io_dime: %17: i1) -> (io_dispense: i1)
```
And the debug information is not being transferred from the `connect` ops to the `hw.instance` op which replaces them.

Try adding the `DontTouch` annotation to some aggregate wires and nodes and you will have some more debug information making it all the way to `export-verilog`

P.S a more efficient pattern for moving the debug information is:
```c++
if (auto debugAttr = op->getAttr("hw.debug.name"))
  other->setAttr("hw.debug.name", debugAttr);
```

Kuree, 2022-01-28:
Thanks for the explanation and I will close the issue for now.

closed 2022-01-28 by Kuree

Ramlakshmi3733, 2022-02-08:
Hi Keyi,
I am at SiFive with the group that is building CIRCT (along with the other open source community members).
I would like to get in touch with you to talk about your interests and career.  If you can, please call me at 650 319 5871

Ran Sunder
VP, Platform Technologies
SiFive
1875 S. Grant St., San Mateo, CA