llvm/circt #2538: [FIRRTL] unable to infer width with known input width

sequencer, 2022-01-30:
Here is the input MLIR:
```
firrtl.circuit "Example" {
  firrtl.module @Example(in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>, in %in: !firrtl.uint<2>, out %out: !firrtl.uint<2>) {
    %_GEN_0 = firrtl.add %in, %in : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint
    %_out_T = firrtl.node %_GEN_0 : !firrtl.uint
    %_GEN_1 = firrtl.add %in, %in : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint
    %_GEN_2 = firrtl.tail %_out_T, 1 : (!firrtl.uint) -> !firrtl.uint
    %_out_T_1 = firrtl.node %_GEN_2 : !firrtl.uint
    %_GEN_3 = firrtl.tail %_out_T, 1 : (!firrtl.uint) -> !firrtl.uint
    firrtl.connect %out, %_out_T_1 : !firrtl.uint<2>, !firrtl.uint
  }
}
```
Since CIRCT doesn't have unknown type representation.(but it has unknown width), for each `PrimOp`, I inferred the return type of which, but left width unknown.
When `firtool` parsing this result, `firtools` complains incompatible type and refuse to parse.
```
<stdin>:3:15: error: 'firrtl.add' op inferred type(s) '!firrtl.uint<3>' are incompatible with return type(s) of operation '!firrtl.uint'
    %_GEN_0 = firrtl.add %in, %in : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint
              ^
<stdin>:3:15: note: see current operation: %0 = "firrtl.add"(%arg2, %arg2) : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint
```
I wonder if is possible to let MFC InferType automatically override unknown width and only show incompatible error on real incompatible type?

seldridge, 2022-01-31:
I'm hesitant to allow this... My reasoning is that the IR must always be in a valid state. Pedantically, the IR above _is invalid_ and relying on a pass to fix it later feels like an anti-pattern (though an anti-pattern that the SFC uses frequently 🥲). This is what `InferTypeOpInterface` is supposed to be for. It allows you to create an `add` without having to specify the result type and instead relying on this being correctly generated based on the operand types.

If we go down this route, we add an additional return type that has to be reasoned about and it means that more passes have to be able to handle unknown widths (and potentially if they see it, it's a pass scheduling error where width inference needs to be run again).

Could you provide more justification on what the motivation is here (generating MLIR text without having to infer widths?)?

sequencer, 2022-01-31:
> Could you provide more justification on what the motivation is here (generating MLIR text without having to infer widths?)?

My intention was proving a semantically correct but as primitive as possible MLIR directly from SFC(I thought this is valid, since it can be parsed), to eliminate the dependency to current SFC pass.

> My reasoning is that the IR must always be in a valid state.

Now I think my question becomes: what's the valid state of a FIRRTL MLIR?
I saw MFC do support unknown width, so I didn't infer width in the SFC pass, since I thought an unknown width of MLIR is valid as well.

> Pedantically, the IR above is invalid and relying on a pass to fix it later feels like an anti-pattern (though an anti-pattern that the SFC uses frequently 🥲). This is what InferTypeOpInterface is [...]

I do agree that is strange LOL. I really like the design of `InferTypeOpInterface` in MFC, I think this is the best way to resolve issue like this.

> If we go down this route, we add an additional return type that has to be reasoned about and it means that more passes have to be able to handle unknown widths (and potentially if they see it, it's a pass scheduling error where width inference needs to be run again).

Since unknown width with uint/sint/analog is valid semantically, so maybe when parsing a MLIR file, it can exist unknown type but after infer type pass, all pass should assume there won't be any unknown width anymore.  

So my propose is: when parsing a MLIR, this file can be "minimal valid", which is just semantically correct, then after infer width and other passes, it becomes more concrete, then other pass can assume they are not possible to generate a non-concrete result again.

darthscsi, 2022-01-31:
> My intention was proving a semantically correct but as primitive as possible MLIR directly from SFC(I thought this is valid, since it can be parsed), to eliminate the dependency to current SFC pass.

Parsing includes the validation, which is what is complaining here, so it doesn't actually parse.

> > My reasoning is that the IR must always be in a valid state.
> 
> Now I think my question becomes: what's the valid state of a FIRRTL MLIR? I saw MFC do support unknown width, so I didn't infer width in the SFC pass, since I thought an unknown width of MLIR is valid as well.

Return types are always explicit.  To have an un-widthed return, you need at least one un-widthed argument.  You know (table in 7.1 of the firrtl spec) the width of the add, so the IR explicitly has that type.  It only doesn't have that type if you don't know the width of the add from the arguments.

> > Pedantically, the IR above is invalid and relying on a pass to fix it later feels like an anti-pattern (though an anti-pattern that the SFC uses frequently smiling_face_with_tear). This is what InferTypeOpInterface is supposed to be for. It allows you to create an add without having to specify the result type and instead relying on this being correctly generated based on the operand types.
> 
> I do agree that is strange LOL. I really like the design of `InferTypeOpInterface` in MFC, I think this is the best way to resolve issue like this.

Eventually, the IR for add should just specify the argument types (e.g. int<2>,  int<2>), since the return type is unique given the argument types.

> Since unknown width with uint/sint/analog is valid semantically, so maybe when parsing a MLIR file, it can exist unknown type but after infer type pass, all pass should assume there won't be any [...]

> So my propose is: when parsing a MLIR, this file can be "minimal valid", which is just semantically correct, then after infer width and other passes, it becomes more concrete, then other pass can assume they are not possible to generate a non-concrete result again.

We don't gain anything, and introduce bug opportunities and complexity, by allowing types besides those uniquely required by an operation in the operation. There is no reason to not give the proper type of the expression in this example.  There is value in reducing the number things needed to be specified, when they are uniquely determined, which what I think you really want here.  Reusing FIRRTL uninfered-width types to do so is the wrong way to achieve this goal.

sequencer, 2022-01-31:
> You know (table in 7.1 of the firrtl spec) the width of the add, so the IR explicitly has that type. It only doesn't have that type if you don't know the width of the add from the arguments.

Sound reasonable, and that do answer the my question!

> There is value in reducing the number things needed to be specified, when they are uniquely determined, which what I think you really want here.

Yes. Having less information do help simplify frontend tools by not doing too much computations. I also saw this in the [discourse thread](https://llvm.discourse.group/t/type-inference-in-rtl/2438/5). I think that will essentially resolve this issue.

Do my only question is, is there some definition to FIRRTL MLIR to define the valid serialization for representation? For example:
1. In FIRRTL MLIR file, is unknown type valid?
2. In FIRRTL MLIR file, is unknown width valid? 

I would like to follow this to emit a FIRRTL MLIR dialect.

seldridge, 2023-11-21:
I think this is resolved as this is definitely illegal MLIR. Most of the issues around serializing to MLIR are handled with the binder effort on Chisel.

closed 2023-11-21 by seldridge