llvm/circt #2561: [Handshake] Integration tests broken recently.

mikeurbach, 2022-02-01:
@mortbopet we don't test these in CI (requires a commercial simulator), but the Handshake integration tests are currently broken, probably by 5fc6ef49b78883890efa1798693d42546294bea4. The failure I'm seeing is:

```
error: 'handshake.buffer' op result 0 has multiple uses.
```

Both tests fail in this way. Let me know if you can see offhand why this fails, or else I'll take a look at it.

mortbopet, 2022-02-01:
Looks like an issue with forks/sinks not being materialized after buffer insertion. I'll push a quick fix (just add materialization) but will look into why this popped up in the first place.

mikeurbach, 2022-02-01:
Thanks, that fixes the tests for me. Was hoping it was something like this.

closed 2022-02-01 by mikeurbach