llvm/circt #2565: [FIRRTL] Simplify PrefixModules pass

prithayan, 2022-02-02:
`PrefixModules` is a global pass that renames the modules based on certain annotation.
 It seems to be a relatively expensive pass for just renaming the modules.
Can we simplify the pass for certain cases, like avoiding cloning modules, when each module is instantiated only once. 
Or, perform it using `verilogName` attribute during `LowerToHW` ?
Some interesting discussion in the original PR: https://github.com/llvm/circt/pull/1183
This is a snapshot of runtime for an internal test case (`p550`) for the FIRRTL pipeline.

```
===-------------------------------------------------------------------------===
                         ... Execution time report ...
===-------------------------------------------------------------------------===
  Total Execution Time: 264.1102 seconds

  ----User Time----  ----Wall Time----  ----Name----
   32.3520 (  5.9%)   32.3520 ( 12.2%)  FIR Parser
  476.1045 ( 86.8%)  191.6725 ( 72.6%)  'firrtl.circuit' Pipeline
   62.5173 ( 11.4%)    3.7192 (  1.4%)    'firrtl.module' Pipeline
   57.9840 ( 10.6%)    3.5445 (  1.3%)      CSE
    0.1501 (  0.0%)    0.0231 (  0.0%)        (A) DominanceInfo
    4.2898 (  0.8%)    0.3141 (  0.1%)      LowerCHIRRTLPass
   11.0799 (  2.0%)   11.0799 (  4.2%)    InferWidths
   12.4964 (  2.3%)   12.4964 (  4.7%)    InferResets
    0.6812 (  0.1%)    0.6812 (  0.3%)      (A) circt::firrtl::InstanceGraph
    5.1985 (  0.9%)    5.1985 (  2.0%)    WireDFT
   11.0132 (  2.0%)   11.0132 (  4.2%)    PrefixModules
   28.7032 (  5.2%)   28.7032 ( 10.9%)    LowerFIRRTLTypes
  179.6644 ( 32.8%)   13.8885 (  5.3%)    'firrtl.module' Pipeline
   32.1330 (  5.9%)    4.2363 (  1.6%)      ExpandWhens
    0.3756 (  0.1%)    0.1330 (  0.1%)      RemoveResets
  135.7015 ( 24.7%)   11.2296 (  4.3%)      Canonicalizer
   11.3954 (  2.1%)    0.8129 (  0.3%)      InferReadWrite
    9.1897 (  1.7%)    9.1897 (  3.5%)    Inliner
   15.1846 (  2.8%)   15.1846 (  5.7%)    IMConstProp
    0.4841 (  0.1%)    0.4841 (  0.2%)      (A) circt::firrtl::InstanceGraph
    3.3466 (  0.6%)    3.3466 (  1.3%)    BlackBoxReader
   48.4989 (  8.8%)   48.4989 ( 18.4%)    GrandCentral
    0.4658 (  0.1%)    0.4658 (  0.2%)      (A) circt::firrtl::InstanceGraph
    6.8357 (  1.2%)    6.8357 (  2.6%)    GrandCentralTaps
    0.5108 (  0.1%)    0.5108 (  0.2%)      (A) circt::firrtl::InstanceGraph
   47.6537 (  8.7%)    7.5269 (  2.8%)    'firrtl.module' Pipeline
    0.2128 (  0.0%)    0.0234 (  0.0%)      GrandCentralSignalMappings
   47.2536 (  8.6%)    7.5137 (  2.8%)      Canonicalizer
    5.5339 (  1.0%)    5.5339 (  2.1%)    RemoveUnusedPorts
    0.4769 (  0.1%)    0.4769 (  0.2%)      (A) circt::firrtl::InstanceGraph
    3.3695 (  0.6%)    3.3695 (  1.3%)    CreateSiFiveMetadata
    0.4371 (  0.1%)    0.4371 (  0.2%)      (A) circt::firrtl::InstanceGraph
    6.0398 (  1.1%)    6.0398 (  2.3%)    EmitOMIR
   39.6520 (  7.2%)   39.6520 ( 15.0%)  Print .mlir output
    0.4314 (  0.1%)    0.4314 (  0.2%)  Rest
  548.5422 (100.0%)  264.1102 (100.0%)  Total
```

youngar, 2022-02-02:
I agree that the pass seems very complicated for just throwing a prefix on to each module's name.  I think the original SFC pass has much more advanced functionality than we need or use.  From what I know, we only use this to put a single prefix to every module in the DUT, there is no composing of multiple prefixes or module cloning taking place. It would be fantastic if we could find some other way simplify the pass or just rip out some functionality.

It might be worth profiling why exactly its taking so much time: the pass [skips cloning](https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Transforms/PrefixModules.cpp#L224) a module if there is only 1 prefix for the module

One thing that might make it simpler is to process a single a `PrefixNestedModules` pass at a time.  It would be simpler to handle a single prefix at a time instead doing them all in a single pass. It would make handling multiple prefixes slower (I think) but we only ever use 1 prefix anyways.

mwachs5, 2022-02-02:
> rom what I know, we only use this to put a single prefix to every module in the DUT, there is no composing of multiple prefixes

False. Multiple nested prefixes are used

seldridge, 2026-08-31:
This pass was removed in: https://github.com/llvm/circt/pull/7944

closed 2026-08-31 by seldridge