llvm/circt #2576: [ExportVerilog] Inline extract op more aggressively

uenoku, 2022-02-03:
For the following ir, 
```
hw.module @Foo(%a: i2) -> (h:i1, j:i1) {
  %a0 = comb.extract %a from 0 : (i2) -> i1
  hw.output %a0, %a0 : i1, i1
}

```
The result is:
```
module Foo(	// bar.mlir:1:1
  input  [1:0] a,
  output       h, j);

  wire _a_0 = a[0];	// bar.mlir:2:9
  assign h = _a_0;	// bar.mlir:3:3
  assign j = _a_0;	// bar.mlir:3:3
endmodule
```
We can reduce number of temporary wires by inlining a[0].
```
assign h = a[0];
assign j = a[0];
```

uenoku, 2022-02-03:
https://github.com/llvm/circt/commit/6d4dcd3c6b872ecefe94b42adf2ff833931a8774 
Seems related to this.

uenoku, 2022-02-04:
This also returns a similar result
```mlir
hw.module @top(%source: !hw.array<1xi1>) -> (res:i1, res2:i1) {
    %false = hw.constant false
    %0 = hw.array_get %source[%false] : !hw.array<1xi1>
    hw.output %0, %0 : i1,i1
  }
}
```
```sv
module top(	// out.mlir:2:3
  input  [0:0] source,
  output       res, res2);

  wire _T = source[1'h0];	// out.mlir:3:13, :4:12
  assign res = _T;	// out.mlir:5:5
  assign res2 = _T;	// out.mlir:5:5
endmodule
```

We might want to inline certain operations even when they have multiuses

teqdruid, 2022-02-04:
When we first did export verilog support for array_get, it wasn't immediately clear when it was syntactically valid to inline them so I just punted. Getting more aggressive on this would significantly de-clutter some internal code I'm working on.

uenoku, 2022-02-04:
Yeah, that makes sense. I think we can allow some operations to be inlined regardless of the number of their users if the operations are cheap to inline.

PR #2582 (2022-02-05): [ExportVerilog] Inline duplicatable unary op more even with multiuses
This commit allows extract op and array_get op to be inlined even when
they have multiple uses. To prevent exponential emission, the index of
array_get must be constant.

Close #2576
closed 2022-02-05 by uenoku