llvm/circt #2588: [ExportVerilog] [PrepareForEmission] Invalid wire creation with disallowLocalVariables option

uenoku, 2022-02-06:
```mlir
// circt-opt -export-verilog %s
module attributes {circt.loweringOptions = "disallowLocalVariables,verifLabels,emittedLineLength=240"} {
hw.module @Foo(%r: i1, %f: i1) -> (foo: i1) {
  %false = hw.constant false
  %reg = sv.reg sym @__reg__  : !hw.inout<i1>
  sv.initial {
    %3 = sv.read_inout %1 : !hw.inout<i1>
    %6 = comb.or %f, %3 : i1
    sv.if %3 {
      sv.passign %reg, %r : i1
    }
  }
  %0 = sv.read_inout %reg : !hw.inout<i1>
  %.io_writeVec.wire = sv.wire  : !hw.inout<array<1xi1>>
  %1 = sv.array_index_inout %.io_writeVec.wire[%false] : !hw.inout<array<1xi1>>, i1
  hw.output %0 : i1
}
}
```

```
<unknown>:0: error: invalid element for hw.inout type '!hw.inout<i1>'
circt-opt: ../llvm/llvm/../mlir/include/mlir/IR/StorageUniquerSupport.h:140: static ConcreteT mlir::detail::StorageUserBase<circt::hw::InOutType, mlir::Type, circt::hw::detail::InOutTypeStorage, mlir::detail::TypeUniquer>::get(mlir::MLIRContext *, Args...) [ConcreteT = circt::hw::InOutType, BaseT = mlir::Type, StorageT = circt::hw::detail::InOutTypeStorage, UniquerT = mlir::detail::TypeUniquer, Traits = <>, Args = <mlir::Type>]: Assertion `succeeded(ConcreteT::verify(getDefaultDiagnosticEmitFn(ctx), args...))' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ./bin/circt-opt -export-verilog ex.mlir
 #0 0x00000000005b2193 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/uenoku/dev/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:13
 #1 0x00000000005b02d0 llvm::sys::RunSignalHandlers() /home/uenoku/dev/circt/llvm/llvm/lib/Support/Signals.cpp:97:18
 #2 0x00000000005b27ca SignalHandler(int) /home/uenoku/dev/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #3 0x00007f4efc0901f0 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x141f0)
 #4 0x00007f4efbaeffbb raise ./signal/../sysdeps/unix/sysv/linux/raise.c:50:1
 #5 0x00007f4efbad5864 abort ./stdlib/abort.c:81:7
 #6 0x00007f4efbad5749 get_sysdep_segment_value ./intl/loadmsgcat.c:509:8
 #7 0x00007f4efbad5749 _nl_load_domain ./intl/loadmsgcat.c:970:34
 #8 0x00007f4efbae73d6 (/lib/x86_64-linux-gnu/libc.so.6+0x383d6)
 #9 0x00000000009bda70 mlir::detail::StorageUserBase<circt::hw::InOutType, mlir::Type, circt::hw::detail::InOutTypeStorage, mlir::detail::TypeUniquer>::getImpl() const /home/uenoku/dev/circt/build/../llvm/llvm/../mlir/include/mlir/IR/StorageUniquerSupport.h:184:68
#10 0x00000000009bda70 circt::hw::InOutType::getElementType() const /home/uenoku/dev/circt/build/tools/circt/include/circt/Dialect/HW/HWTypes.cpp.inc:172:10
#11 0x0000000000a5a6c8 circt::sv::WireOp::build(mlir::OpBuilder&, mlir::OperationState&, mlir::Type, mlir::StringAttr, mlir::StringAttr) /home/uenoku/dev/circt/lib/Dialect/SV/SVOps.cpp:1010:21
#12 0x00000000006bd41a circt::sv::WireOp mlir::OpBuilder::create<circt::sv::WireOp, mlir::Type>(mlir::Location, mlir::Type&&) /home/uenoku/dev/circt/build/../llvm/llvm/../mlir/include/mlir/IR/Builders.h:431:16
#13 0x00000000006bbce7 lowerUsersToTemporaryWire(mlir::Operation&) /home/uenoku/dev/circt/lib/Conversion/ExportVerilog/PrepareForEmission.cpp:229:28
#14 0x00000000006bbce7 circt::ExportVerilog::prepareHWModule(mlir::Block&, circt::LoweringOptions const&) /home/uenoku/dev/circt/lib/Conversion/ExportVerilog/PrepareForEmission.cpp:541:7
#15 0x00000000006941ed mlir::LogicalResult mlir::failableParallelForEach<circt::hw::HWModuleOp*, void mlir::parallelForEach<circt::hw::HWModuleOp*, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71>(mlir::MLIRContext*, circt::hw::HWModuleOp*, circt::hw::HWModuleOp*, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71&&)::'lambda'(circt::hw::HWModuleOp*&&)>(mlir::MLIRContext*, circt::hw::HWModuleOp*, circt::hw::HWModuleOp*, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71&&) /home/uenoku/dev/circt/build/../llvm/llvm/../mlir/include/mlir/IR/Threading.h:48:26
#16 0x00000000006941ed void mlir::parallelForEach<circt::hw::HWModuleOp*, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71>(mlir::MLIRContext*, circt::hw::HWModuleOp*, circt::hw::HWModuleOp*, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71&&) /home/uenoku/dev/circt/build/../llvm/llvm/../mlir/include/mlir/IR/Threading.h:124:9
#17 0x00000000006941ed void mlir::parallelForEach<llvm::SmallVector<circt::hw::HWModuleOp, 6u>&, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71>(mlir::MLIRContext*, llvm::SmallVector<circt::hw::HWModuleOp, 6u>&, prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&)::$_71&&) /home/uenoku/dev/circt/build/../llvm/llvm/../mlir/include/mlir/IR/Threading.h:136:3
#18 0x00000000006941ed prepareForEmission(mlir::ModuleOp, circt::LoweringOptions const&) /home/uenoku/dev/circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:4370:3
```

Seems prepareForEmission tries to create temporary wires for inout operations.

mikeurbach, 2022-09-30:
I was working on another change that is tickling this behavior. I will try to resolve this since I am pretty sure my change is legal and I'd like to be able to use it.

mikeurbach, 2022-09-30:
Seems the original test case doesn't crash, but you can still tickle this with InstanceOps that have inout operands. My change was accidentally creating inouts, which is legal, but not what I intended. So a testcase for this would be an instance with an operand like `"foo": %wire: !hw.inout<i1>`.

uenoku, 2022-09-30:
Probably we can close this now since the original issue was fixed.  I opened a new issue for crash when inout o ports https://github.com/llvm/circt/issues/4036

closed 2022-09-30 by uenoku