llvm/circt #2590: [ExportVerilog] Unnecessary temporary wire for bitcast between array and integer

uenoku, 2022-02-06:
input: 
```mlir
hw.module @Foo(%a: i2) -> (r:i1) {
  %0 = hw.bitcast %a : (i2) -> !hw.array<2xi1>
  %false = hw.constant false
  %1 = hw.array_get %0[%false] : !hw.array<2xi1>
  hw.output %1 : i1
}
```
output :
```sv
module Foo(	// foo.mlir:1:1
  input  [1:0] a,
  output       r);

  wire [1:0] _T = a;	// foo.mlir:2:8
  assign r = _T[1'h0];	// foo.mlir:3:11, :4:7, :5:2
endmodule
```

This pattern would not be so frequent but we can eliminate `_T`.

teqdruid, 2022-02-06:
Yeah, I wasn't sure where it was legal to cast. Not so much a problem for arrays (I think) but once one gets into structs... I suspect the verilog rules regarding casts are bizarre.

uenoku, 2022-02-07:
Yes, I agree inlining bitcast should be illegal in general but for bitcast between array<2xi1> and i2, it is valid