llvm/circt #2593: [ExportVerilog] Omit bitwidth of constant array index

uenoku, 2022-02-07:
```mlir
hw.module @Foo(%a: !hw.array<1xi1>) -> (b: i1) {
  %false = hw.constant false
  %0 = hw.array_get %a[%false] : !hw.array<1xi1>
  hw.output %0 : i1
}
```

```sv
module Foo(	// foo.fir:2:10
  input  [0:0] a,
  output       b);

  assign b = a[1'h0];	// foo.fir:2:10, :5:12
endmodule
```

I'm not 100% sure it is always legal but can't we just emit `assign b = a[0]` ?

teqdruid, 2022-02-07:
Yes, it is almost certainly legal in all cases. I do recall some investigation of what verilator and questa support in terms of array index bitwidths, though I can't find it anymore and I don't think it was around constant indexes.

uenoku, 2022-02-08:
Thanks, I agree it should be legal in most cases but seems problematic when the constant is larger than 32-bit integer. (Probably we can omit bitwidth for most of constants, but I may have to investigate the spec)