llvm/circt #2624: [ExportVerilog] Reuse wire/register declarations

uenoku, 2022-02-14:
```mlir
; circt-opt -export-verilog
  hw.module @Foo() {
    %0 = sv.wire  : !hw.inout<i3>
    %1 = sv.wire  : !hw.inout<i3>
    hw.output
  }
```
Currently the output is:
```sv
module Foo();  
  wire [2:0] _GEN;  
  wire [2:0] _GEN_0;  

endmodule
``` 

Ideally it would be better to reuse declaration of wires like port delcrations:
```sv
module Foo(); 
  wire [2:0] _GEN, _GEN_0;   

endmodule
```

sinofp, 2022-02-14:
I'm not a Verilog master, but I think the current output is better because I only need to read in one direction.

uenoku, 2022-02-14:
> I'm not a Verilog master, but I think the current output is better because I only need to read in one direction.

That's a very reasonable point. I was considering this by looking at huge amount of lines of wire declarations.  
(For example, the current output of `DigitalTop` module in chipyard benchmark has 700 lines of wire declarations:
https://gist.github.com/uenoku/13586b849977d668790113dc188911b0#file-chipyard_digital_top-sv-L91-L784)