llvm/circt #2660: [Calyx] SCFToCalyx generates zero-width memref address ports for memrefs with some dimension = 1

makslevental, 2022-02-20:
`memref.store %12, %2[%arg1, %arg2, %arg3, %arg4] : memref<1x3x9x9xi32>`

produces stuff like 

`calyx.assign %mem_2.addr0 = %std_slice_20.out : i0`

Basically due to `log2(1) = 0` ([here](https://github.com/llvm/circt/blob/main/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L1086) and [here](https://github.com/llvm/circt/blob/main/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L1456)).

Easy fix. Going through the steps to submit a patch through fabricator.

mortbopet, 2022-02-20:
So in some sense, this isn't necessarily an error; zero width signals have been discussed previously in the ODMs (https://docs.google.com/document/d/1fOSRdyZR2w75D87yU2Ma9h2-_lEPL4NxvhJGJd-s5pk/edit#heading=h.8tkap0za1yar march 17 2021).
In a memory address context, zero-width signals makes sense in cases where a memory _might_ be a singleton, as you've identified, but where we _don't_ want to add special-case logic all over the place to handle this in our compiler.

The issue here is then that we need to export this to the Calyx native compiler, and so this needs to be resolved (@rachitnigam does calyx support zero-width signals?).
If/when we keep things in CIRCT, i'd expect this kind of thing to be legal, and dealt with further down the chain, at the RTL level.

As a temporary solution, i think the best way around this is to ensure that the [`--flatten-memref` pass](https://github.com/llvm/circt/blob/33c4c645b88222104478eadd88adb5af7b4db932/include/circt/Transforms/Passes.td#L19) is called prior to `--scf-to-calyx`. This would be in line with the [discussion going on in Calyx about possibly removing multidimensional memories from the core primitives](https://github.com/cucapra/calyx/issues/907).

uenoku, 2022-02-20:
Zero-width values at HW dialect have been in demand several times: https://github.com/llvm/circt/issues/2219. 
Originally it was due to limitation of APInt but that limitation was resolved so it might be good opportunity to support zero-width in HW dialect (it will introduce more complexity to comb foldings and ExportVerilog though).

rachitnigam, 2022-02-20:
Calyx does not directly support zero-width signals because of the aforementioned problems with specializing things in various optimizations. However, there is no strong stance for or against the idea currently so happy to discuss it!

closed 2022-03-17 by mikeurbach