llvm/circt #2677: [Handshake] Handshake dialect missing from CIRCT docs

mortbopet, 2022-02-24:
As seen in https://circt.llvm.org/docs/Dialects/, the Handshake dialect seems to be missing from the autogenerated documentation.

wizvivek, 2022-02-26:
If i am not wrong, i will be provided with material and i just have to document it.

mortbopet, 2022-02-26:
@wizvivek the dialect documentation should be autogenerated from the dialect TableGen files - so, somewhere that autogenerated documentation is not being included. I took another quick look at this and stumbled across this, wherein we were only generating operation-level docs for the Handshake dialect, and not dialect-level docs!

https://github.com/llvm/circt/blob/106815aea4fe252f270af796a1b1762c77c15b0d/include/circt/Dialect/Handshake/CMakeLists.txt#L2

Fixed in PR #2697

closed 2022-02-26 by mortbopet