llvm/circt #2685: Windows build broken

youngar, 2022-02-24:


teqdruid, 2022-02-24:
Since the bump compiled successfully, I suspect a build environment change.
I'm re-running the last bump build to check: https://github.com/llvm/circt/actions/runs/1891200560

teqdruid, 2022-02-24:
Yep. Build environment change.

teqdruid, 2022-02-24:
I _suspect_ we got bumped to server 2022. Test build with downgrade: https://github.com/llvm/circt/actions/runs/1895364998

teqdruid, 2022-02-24:
Made it through config, so yeah. Might not have been bumped, but downgrading fixed it.

closed 2022-02-24 by teqdruid

youngar, 2022-02-24:
I'm just noticing it says we were bumped in the Summary page:
[image]

youngar, 2022-02-24:
also, thanks for fixing this! 😁

teqdruid, 2022-02-24:
Serves me right for not reading the warnings!

commit 85a3a96b (2022-02-24): [Windows CI] Downgrade agent os version

Fixes #2685