llvm/circt #2691: [FIRRTL][GrandCentralTap] GCTaps doesn't understand non-local annotations produced by dedup

youngar, 2022-02-25:
In the following example, we have a similar module `Foo` and `Bar`.  `Foo` has a data tap on it.  `Bar` is later deduplicated with `Foo` and removed.  Some time after this, Tap selects the wrong instance, `bar of Bar` for the DataTap.
```
circuit Top : %[[{
  "class": "sifive.enterprise.grandcentral.DataTapsAnnotation",
  "blackBox": "~Top|DataTap",
  "keys": [
    {
      "class": "sifive.enterprise.grandcentral.ReferenceDataTapKey",
      "source": "~Top|Bar>b",
      "portName": "~Top|DataTap>b"
    }
]}]]
  extmodule DataTap :
    output b : UInt<1>
    defname = DataTap
  module Foo :
    wire f : UInt<1>
    f is invalid
  module Bar :
    wire b : UInt<1>
    b is invalid
  module Top:
    inst foo of Foo
    inst bar of Bar
    inst dataTap of DataTap
```
`./bin/firtool -firrtl-grand-central --dedup ./gc.fir`
```
module DataTap_impl_0(  // ./gc.fir:11:13
  output b);

  assign b = Top.bar.f; // ./gc.fir:11:13
endmodule

module Foo();   // ./gc.fir:14:10, :17:10
  wire f;       // ./gc.fir:15:5, :18:5

  assign f = 1'h0;      // ./gc.fir:16:7, :19:7
endmodule

module Top();   // ./gc.fir:20:10
  wire dataTap_b;       // ./gc.fir:23:5

  Foo foo ();   // ./gc.fir:21:5
  Foo bar ();   // ./gc.fir:22:5
  DataTap_impl_0 dataTap (      // ./gc.fir:23:5
    .b (dataTap_b)
  );
endmodule
```

fabianschuiki, 2022-02-25:
I'm pretty sure the Grand Central Taps pass doesn't properly process NLAs. I remember that I had to jump through a few hoops to figure out which instance a `ReferenceDataTapKey` was referring to (because they were all targeting modules, but actually meant instances). There's some logic in the pass implementation that looks at all instantiations of modules and picks the one that is "most likely the intended one". If I recall correctly, that's just picking the shortest hierarchical path or something like that. Now that we have NLAs, that part of the code should probably just check if there's an NLA and use that as the precise locator (instead of trying to cook something up ad-hoc).

fabianschuiki, 2022-02-25:
Running with `--debug-only=gct` produces:
```
Generating impls for DataTap
- $root/dataTap:DataTap
- Processing port 0 anno {class = "sifive.enterprise.grandcentral.ReferenceDataTapKey", id = 0 : i64, portID = 1 : i64, type = "portName"}
- Wire up as follows:
- Port 0:
  - $root/bar:Foo.
  - $root/foo:Foo.
Implementing "DataTap_impl_0" (DataTap for $root/dataTap:DataTap)
- Wiring up port 0
  - Shortest prefix $root/bar:Foo
  - Connecting as {{0}}.{{1}}.{{2}}
```
That "shortest prefix `$root/bar:Foo`" is the culprit.

closed 2022-02-25 by fabianschuiki