llvm/circt #2696: [firtool] --verbose-pass-executions should print parser info

seldridge, 2022-02-25:
Currently when you run `firtool --verbose-pass-executions`, it doesn't print any information when it's in the parser.  This can be confusing to a user especially for large designs where, currently, >50% of the time is spent in the parser.

Modify `--verbose-pass-executions` to print information about parsing.

Sample output:

```
# firtool Foo.fir -verbose-pass-executions -mlir-timing
[firtool] Running "firrtl.circuit(firrtl.module(cse, firrtl-lower-chirrtl), firrtl-infer-widths, firrtl-infer-resets, firrtl-dft, firrtl-prefix-modules, firrtl-lower-types{flatten-mem=false preserve-aggregate=false preserve-public-types=true}, firrtl.module(firrtl-expand-whens, firrtl-remove-resets, canonicalize{  max-iterations=10 region-simplify=true top-down=true}, firrtl-infer-rw), firrtl-inliner, firrtl-imconstprop, firrtl-blackbox-reader{input-prefix= resource-prefix=}, firrtl.module(canonicalize{  max-iterations=10 region-simplify=true top-down=true}), firrtl-remove-unused-ports, firrtl-emit-metadata{repl-seq-mem=false repl-seq-mem-circuit= repl-seq-mem-file=}, firrtl-emit-omir{file=})"
[firtool]   Running "firrtl.module(cse, firrtl-lower-chirrtl)"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-infer-widths"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-infer-resets"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-dft"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-prefix-modules"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-lower-types{flatten-mem=false preserve-aggregate=false preserve-public-types=true}"
[firtool]   -- Done in 0.001 sec
[firtool]   Running "firrtl.module(firrtl-expand-whens, firrtl-remove-resets, canonicalize{  max-iterations=10 region-simplify=true top-down=true}, firrtl-infer-rw)"
[firtool]   -- Done in 0.001 sec
[firtool]   Running "firrtl-inliner"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-imconstprop"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-blackbox-reader{input-prefix= resource-prefix=}"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl.module(canonicalize{  max-iterations=10 region-simplify=true top-down=true})"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-remove-unused-ports"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-emit-metadata{repl-seq-mem=false repl-seq-mem-circuit= repl-seq-mem-file=}"
[firtool]   -- Done in 0.000 sec
[firtool]   Running "firrtl-emit-omir{file=}"
[firtool]   -- Done in 0.000 sec
[firtool] -- Done in 0.006 sec
[firtool] Running "lower-firrtl-to-hw{error-on-non-const-async-reset-values=true warn-on-unprocessed-annotations=false}"
[firtool] -- Done in 0.001 sec
[firtool] Running "hw-memory-sim{ignore-read-enable-mem=false repl-seq-mem=false}"
[firtool] -- Done in 0.000 sec
[firtool] Running "hw.module(cse, canonicalize{  max-iterations=10 region-simplify=true top-down=true}, hw-cleanup, hw-legalize-modules, prettify-verilog)"
[firtool] -- Done in 0.001 sec
[firtool] Running "export-verilog"
```

uenoku, 2022-02-26:
firtool prints logs of execution when ` --verbose-pass-executions` option is enabled. Currently start/end messages are not printed when parsing (because mlir/fir parser is not executed on mlir pass manager).  Then we want to print something like this:  
```
$ firtool Foo.fir -verbose-pass-executions -mlir-timing
[firtool] Running parser
[firtool] -- Done in 0.001 sec
...
```

I think the code we want to change is around following part:
https://github.com/llvm/circt/blob/main/tools/firtool/firtool.cpp#L381-L405

wizvivek, 2022-02-26:
sorry, i'm unable to do this task.

closed 2022-04-04 by uenoku