llvm/circt #2704: [Tracker] Handshake-to-HW lowering

mortbopet, 2022-03-01:
This issue intends to track the progress on completing the [handshake-to-hw lowering](https://github.com/llvm/circt/blob/main/include/circt/Conversion/HandshakeToHW.h), and by doing so, transition off of our current [FIRRTL-based lowering path](https://github.com/llvm/circt/blob/main/include/circt/Conversion/HandshakeToFIRRTL.h).

The 

* [x] implement top-level lowering c7c07659335580fc531ac58e840be056b8c8fcd8
* [ ] Implement generic versions of handshake operations
  * In the FIRRTL flow, we algorithmically describe each handshake operator using C++ logic. This is fairly cumbersome, since operators are generic wrt. their input types and # of input arguments. Instead, we would like to have these modules expressed as generic HW modules, which can be specialized on instantiation.
  * To support this, we still need some capabilities at the HW level. Most notably, we need to have a form of `generate` statement that can elaborate a module based on input parameters.
* [ ] Determine a memory abstraction
  * FIRRTL provides a memory operation that greatly simplifies lowering memories in the current flow.  The `seq` dialect does not yet have such memory abstraction. To transition, we therefore need to figure out whether to implement first-class support in the `seq` dialect, or if some other route is needed.
* [ ] Determine a top-level memory interface
  * In this new lowering flow, all ground types (data signals) are modeled using ESI channels (ready, valid, data signals). This leaves us with the question of how to expose a memory connection at the top-level (and internally, in the hardware module). Designing around ESI interfaces ideally allows us to compose different hardware modules, and so we'd equally want a memory interface abstraction from ESI.

mikeurbach, 2022-03-03:
Regarding `generate`, I understand the desire to use this functionality, but it seems that will require more thinking. Is it possible to alleviate some of the pain with helpers in C++? Rather than defining `generate` features in the IR, how hard would it be to have builder methods for each Handshake component that are generic w.r.t. their input/output types at the C++ level?

I'm all for defining `generate` at the HW level, but this seems like it will require design discussions etc. to get right, and it might be quicker to make progress on Handshake to HW without it.

mortbopet, 2022-03-05:
> Regarding generate, I understand the desire to use this functionality, but it seems that will require more thinking. Is it possible to alleviate some of the pain with helpers in C++? Rather than [...]

Unless I'm misunderstanding your proposal, I don't think such a solution would look that much different than what we already have today in handshake-to-firrtl.

I'll be experimenting a bit with this generate concept locally and present my thoughts/findings at a later ODM. Alongside this, it's exposing quite a bit of missing stuff from the current state of parametric modules which nevertheless should be good contributions.

mikeurbach, 2022-03-08:
> I don't think such a solution would look that much different than what we already have today in handshake-to-firrtl

That is correct. I'm excited to see what you uncover by pushing on this.