llvm/circt #2705: [HW] Add support for generator expressions

mortbopet, 2022-03-01:
#2701, #2703 introduces support for using HW module parameters in `seq, comb` operations as well as support for parametric arrays.

To be able to do something meaningful with this, a next step could be to add a `generate` operation to the HW dialect, wherein we can use control flow operations to describe our parametric hardware.
This would then be accompanied by an elaboration step that is invoked upon module instantiation. By doing this in CIRCT (as opposed to emitting SystemVerilog `generate` structures), we'll be able to leverage existing RTL dialect optimizations and transformations/inlining/....

To kick off a discussion, one possible implementation is to implement a `hw.generate` operation that defines a region. Internally, `SCF` 
or `affine` dialect operations could be used as the elaboration language. The main challenge with this is to be able to use our `#hw.param.decl.ref<"...">` attributes as values accepted by the verifiers of these operations, such as the following:

```mlir
hw.module<N: i32>(%vec : !hw.array<#hw.param.decl.ref<"N"> x i32>) -> (out: i32) {
  %tmp = sv.wire : i32
  hw.generate {
    %first = hw.array_get %vec[0] : i32
    %sum = scf.for %iv = 1 to #hw.param.decl.ref<"N"> step 1 iter_args(%sum_iter = %first) -> (i32) {
            %v = hw.array_get %vec[%iv] : i32
            %partial_sum = comb.add %sum_iter, %v
            scf.yield %partial_sum : i32
    }
    sv.assign %tmp, %sum : i32
  }
  hw.output %tmp : i32
}
```

darthscsi, 2022-03-01:
It's not clear why we need a generate in the IR.  What semantics of hw.module are unsuitable for directly hosting scf.for or affine or such?  Likewise, given that sv has a for loop, why would we not just be adding an sv.for?

mortbopet, 2022-03-02:
> What semantics of hw.module are unsuitable for directly hosting scf.for or affine or such?

My main concern here is that a `hw.module` implements graph region, whereas i'd see a `hw.generate` operation to implement a SSACFG region.

> Likewise, given that sv has a for loop, why would we not just be adding an sv.for?

By this, do you then imply that this would be emitted as a `for` loop in the emitted SystemVerilog?
As mentioned in the issue, I'd hope that by having elaboration capabilities in CIRCT (as opposed to letting vendor tooling elaborate generated structures), elaborated modules can benefit from all of the optimizations available in the RTL dialects.

mwachs5, 2022-03-09:
> To be able to do something meaningful with this

Couldn't we do something meaningful with this just by doing a parameterized width, before needing to add a generate?