llvm/circt #2707: [firtool] Unused Modules Should be Removed

seldridge, 2022-03-01:
Consider the following:

```scala
circuit Foo:
  module Bar:
    input a: UInt<1>
    output b: UInt<1>

    wire w: UInt<1>

    w <= a
    b <= w

  module Foo:
    output b: UInt<1>

    inst bar of Bar
    bar.a <= UInt<1>(1)
    b <= bar.b
```

When compiled with CIRCT, this leaves module `Bar` around even though it is dead due to constant propagation:

```verilog
module Bar(
  output b);

  wire w;

  assign w = 1'h1;
  assign b = 1'h1;
endmodule

module Foo(
  output b);

  wire bar_b;

  Bar bar (
    .b (bar_b)
  );
  assign b = 1'h1;
endmodule
```

With SFC, `Bar` will be removed:
```verilog
module Foo(
  output  b
);
  assign b = 1'h1;
endmodule
```

uenoku, 2022-03-02:
FYI, this is the dup of https://github.com/llvm/circt/issues/1267

seldridge, 2022-03-02:
Yep! Closing in favor of #1267.