llvm/circt #2709: [FIRRTL][GrandCentralTaps] Module with instance sharing the same name creates a bad NLA

youngar, 2022-03-01:
```
circuit Top : %[[
  {
    "class":"sifive.enterprise.grandcentral.DataTapsAnnotation",
    "blackBox":"~Top|DataTap",
    "keys":[
      {
        "class":"sifive.enterprise.grandcentral.ReferenceDataTapKey",
        "source":"~Top|Top/test:Test/Test:InnerTest>signal",
        "portName":"~Top|DataTap>_0"
      }
    ]
  }
]]
  module Test :
    inst Test of InnerTest
  module InnerTest :
    wire signal : UInt<8>
    signal is invalid
  module Top :
    inst test of Test
    inst DataTap of DataTap
  extmodule DataTap :
    output _0 : UInt<1>
    defname = DataTap
```
When trying to run this test, it creates a bad NLA. The `@Test` FlatSymbolRefAttr should be an `hw.innerNameRef<@Test::@Test>`.
```
./bin/firtool -debug-only=gct -dedup -firrtl-grand-central --strip-debug-info  ./test.fir
./test.fir:1:1: error: 'firrtl.nla' op the instance path can only contain inner sym reference, only the leaf can refer to a module symbol
circuit Top : %[[
^
./test.fir:1:1: note: see current operation: "firrtl.nla"() {namepath = [#hw.innerNameRef<@Top::@test>, @Test, #hw.innerNameRef<@InnerTest::@signal>], sym_name = "nla_1"} : () -> ()
```

darthscsi, 2022-03-01:
There was a similar problem with the symCache here: https://github.com/llvm/circt/commit/3257cb6594ffeccafa07ba38843a00580ec2ff55

youngar, 2022-03-02:
The issue also hits when ports share the same name as the module:
```
  firrtl.circuit "Test"  {
    firrtl.module @Test(in %Test: !firrtl.uint<1> [{class = "fake"}]) {
    }
  }
```
The port reference is changed into a module reference:
```
module {
  firrtl.circuit "Test"  {
    firrtl.nla @nla_1 [#hw.innerNameRef<@Test::@other>, @Other]
    firrtl.module @Test() {
      %other_Other = firrtl.instance other sym @other  {annotations = [{circt.nonlocal = @nla_1, class = "circt.nonlocal"}]} @Other(in Other: !firrtl.uint<1>)
    }
    firrtl.module @Other(in %Other: !firrtl.uint<1> sym @Other [{circt.nonlocal = @nla_1, class = "fake"}]) {
    }
  }
}
```

youngar, 2022-03-02:
this is due to this weird assumption: https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Import/FIRAnnotations.cpp#L148

closed 2022-03-03 by youngar