llvm/circt #2719: [FIRRTL] Dedup access bundle type out-of-bounds element

fabianschuiki, 2022-03-03:
I stumbled across the following bug today when I tried to reduce a test case for a different problem. Basically, when running the `module-externalizer` reduction, which tries to replace as many `firrtl.module` with `firrtl.extmodule` ops as possible, the dedup pass would start to trigger an assertion in `BundleType::getElementType`. I suspect this is somehow related to handling of extmodules, since regular modules didn't trigger the crash.

Here's the reduced test case:
```mlir
firrtl.circuit "Foo"  {
  firrtl.extmodule @Bar(
    in clock: !firrtl.clock, out io: !firrtl.bundle<a: clock>)
  firrtl.extmodule @Baz(
    in clock: !firrtl.clock, out io: !firrtl.bundle<a flip: uint<1>, b flip: uint<16>, c: uint<1>>)
  firrtl.module @Foo() {
    %invalid_ui16 = firrtl.invalidvalue : !firrtl.uint<16>
    %bar_clock, %bar_io = firrtl.instance bar @Bar(
      in clock: !firrtl.clock, out io: !firrtl.bundle<a: clock>)
    %baz_clock, %baz_io = firrtl.instance baz @Baz(
      in clock: !firrtl.clock, out io: !firrtl.bundle<a flip: uint<1>, b flip: uint<16>, c: uint<1>>)
    %0 = firrtl.subfield %baz_io(1)
      : (!firrtl.bundle<a flip: uint<1>, b flip: uint<16>, c: uint<1>>) -> !firrtl.uint<16>
    firrtl.strictconnect %0, %invalid_ui16 : !firrtl.uint<16>
  }
}
```
Running this through dedup produces the following on my machine:
```
# circt-opt --firrtl-dedup
circt-opt: FIRRTLTypes.cpp:843: circt::firrtl::FIRRTLType circt::firrtl::BundleType::getElementType(size_t):
Assertion `index < getNumElements() && "index must be less than number of fields in bundle"' failed.
[...]
Deduper::fixupReferences(mlir::Value, mlir::Type) Dedup.cpp:387:59
Deduper::fixupReferences(circt::firrtl::FModuleLike, mlir::Operation*) Dedup.cpp:449:25
Deduper::dedup(circt::firrtl::FModuleLike, circt::firrtl::FModuleLike) Dedup.cpp:233:3
```

youngar, 2022-03-03:
In this example, it should not be trying to dedup the two extmodules. which is fixed by the linked PR.  There is also the issue that the pass does not support `strictconnect` operations and I opened up a seperate issue for this https://github.com/llvm/circt/issues/2722

fabianschuiki, 2022-03-03:
Ah, good catch about the `strictconnect` thing. I didn't even realize that this was a problem!

closed 2022-03-03 by youngar