llvm/circt #2728: [HandshakeToFIRRTL] Signed Integer Comparison on Signless Integers is lowered incorrectly

Dinistro, 2022-03-04:
I stumbled across an issue when lowering signless integers from handshake to FIRRTL.

Lowering the following handshake code converts the types to unsigned integers.

```mlir
module {
  handshake.func @fuu(%arg0: i32, %arg1: none, ...) -> (i1, none) attributes {argNames = ["in0", "inCtrl"], resNames = ["out0", "outCtrl"]} {
    %0:2 = fork [2] %arg1 : none
    %1 = constant %0#0 {value = 0 : i32} : i32
    %2 = arith.cmpi sgt, %1, %arg0 : i32
    return %2, %0#1 : i1, none
  }
}
``` 

The resulting FIRRTL looks as follows:

```mlir
module {
  firrtl.circuit "fuu"  {
    firrtl.module @handshake_fork_1ins_2outs_ctrl(in %in0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out %out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out %out1: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>) {
      %0 = firrtl.subfield %in0(0) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %1 = firrtl.subfield %in0(1) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %2 = firrtl.subfield %out0(0) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %3 = firrtl.subfield %out0(1) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %4 = firrtl.subfield %out1(0) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %5 = firrtl.subfield %out1(1) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
      %done0 = firrtl.wire  : !firrtl.uint<1>
      %done1 = firrtl.wire  : !firrtl.uint<1>
      %allDone = firrtl.wire  : !firrtl.uint<1>
      %6 = firrtl.and %done1, %done0 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %allDone, %6 : !firrtl.uint<1>, !firrtl.uint<1>
      firrtl.connect %1, %allDone : !firrtl.uint<1>, !firrtl.uint<1>
      %notAllDone = firrtl.wire  : !firrtl.uint<1>
      %7 = firrtl.not %allDone : (!firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %notAllDone, %7 : !firrtl.uint<1>, !firrtl.uint<1>
      %emtd0 = firrtl.regreset %clock, %reset, %c0_ui1  : !firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>
      %8 = firrtl.and %done0, %notAllDone : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %emtd0, %8 : !firrtl.uint<1>, !firrtl.uint<1>
      %notEmtd0 = firrtl.wire  : !firrtl.uint<1>
      %9 = firrtl.not %emtd0 : (!firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %notEmtd0, %9 : !firrtl.uint<1>, !firrtl.uint<1>
      %10 = firrtl.and %notEmtd0, %0 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %2, %10 : !firrtl.uint<1>, !firrtl.uint<1>
      %validReady0 = firrtl.wire  : !firrtl.uint<1>
      %11 = firrtl.and %3, %10 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %validReady0, %11 : !firrtl.uint<1>, !firrtl.uint<1>
      %12 = firrtl.or %validReady0, %emtd0 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %done0, %12 : !firrtl.uint<1>, !firrtl.uint<1>
      %emtd1 = firrtl.regreset %clock, %reset, %c0_ui1  : !firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>
      %13 = firrtl.and %done1, %notAllDone : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %emtd1, %13 : !firrtl.uint<1>, !firrtl.uint<1>
      %notEmtd1 = firrtl.wire  : !firrtl.uint<1>
      %14 = firrtl.not %emtd1 : (!firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %notEmtd1, %14 : !firrtl.uint<1>, !firrtl.uint<1>
      %15 = firrtl.and %notEmtd1, %0 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %4, %15 : !firrtl.uint<1>, !firrtl.uint<1>
      %validReady1 = firrtl.wire  : !firrtl.uint<1>
      %16 = firrtl.and %5, %15 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %validReady1, %16 : !firrtl.uint<1>, !firrtl.uint<1>
      %17 = firrtl.or %validReady1, %emtd1 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %done1, %17 : !firrtl.uint<1>, !firrtl.uint<1>
    }
    firrtl.module @handshake_constant_c0_out_ui32(in %ctrl: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out %out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>) {
      %0 = firrtl.subfield %ctrl(0) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %1 = firrtl.subfield %ctrl(1) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>>) -> !firrtl.uint<1>
      %2 = firrtl.subfield %out0(0) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>) -> !firrtl.uint<1>
      %3 = firrtl.subfield %out0(1) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>) -> !firrtl.uint<1>
      %4 = firrtl.subfield %out0(2) : (!firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>) -> !firrtl.uint<32>
[... 20 lines cut ...]
      firrtl.connect %1, %11 : !firrtl.uint<1>, !firrtl.uint<1>
      firrtl.connect %4, %11 : !firrtl.uint<1>, !firrtl.uint<1>
    }
    firrtl.module @fuu(in %in0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>, in %inCtrl: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out %out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<1>>, out %outCtrl: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>) {
      %handshake_fork0_in0, %handshake_fork0_out0, %handshake_fork0_out1, %handshake_fork0_clock, %handshake_fork0_reset = firrtl.instance handshake_fork0  @handshake_fork_1ins_2outs_ctrl(in in0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out out1: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, in clock: !firrtl.clock, in reset: !firrtl.uint<1>)
      %handshake_constant0_ctrl, %handshake_constant0_out0 = firrtl.instance handshake_constant0  @handshake_constant_c0_out_ui32(in ctrl: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, out out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>)
      %arith_cmpi0_in0, %arith_cmpi0_in1, %arith_cmpi0_out0 = firrtl.instance arith_cmpi0  @arith_cmpi_in_ui32_ui32_out_ui1_sgt(in in0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>, in in1: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>, out out0: !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<1>>)
      firrtl.connect %handshake_fork0_in0, %inCtrl : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>
      firrtl.connect %handshake_fork0_clock, %clock : !firrtl.clock, !firrtl.clock
      firrtl.connect %handshake_fork0_reset, %reset : !firrtl.uint<1>, !firrtl.uint<1>
      firrtl.connect %handshake_constant0_ctrl, %handshake_fork0_out0 : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>
      firrtl.connect %arith_cmpi0_in0, %handshake_constant0_out0 : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>
      firrtl.connect %arith_cmpi0_in1, %in0 : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<32>>
      firrtl.connect %out0, %arith_cmpi0_out0 : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<1>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>, data: uint<1>>
      firrtl.connect %outCtrl, %handshake_fork0_out1 : !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>, !firrtl.bundle<valid: uint<1>, ready flip: uint<1>>
    }
  }
}
```

While this is a bit crowded, one can see that the module `arith_cmpi_in_ui32_ui32_out_ui1_sgt` takes unsigned integers (`uint<32>`), and thus the `firrtl.gt` corresponds to an "unsigned greater than" instead of a "signed greater than" comparison. 

While debugging this issue, I found that the line responsible for type conversion already has an issue comment (https://github.com/llvm/circt/blob/main/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp#L67-L71), but I couldn't find an issue here. 

This problem might be solvable by introducing casts in the form `firrtl.asSInt` and `firrtl.asUInt` before and after the `firrtl.gt`, but I'm not yet too familiar with FIRRTL to be certain.

Dinistro, 2022-03-07:
Similar issues occur for `arith.ceildivsi`, `arith.divsi`, `arith.extsi`, `arith.fptosi`, `arith.floordivsi`, `arith.maxsi`, `arith.minsi`, `arith.remsi`, `arith.sitofp`, and `arith.shrsi`.

According to `LowerFIRRTLToHW`, the `asSInt` operations are replaced by Noops while lowering. Thus inserting them should not be too much of an overhead, or did I miss something?

darthscsi, 2022-03-08:
The casts should be free.  When handshake goes straight to HW, it won't need these casts.

PR #2730 (2022-03-10): [HandshakeToFIRRTL]: Add support for signed ops
This PR adds support for signed operations by casting unsigned operands to signed ones, if needed. 
Closes #2728 as it solves the issues of all the existing operations.  

Note: I decided to template the parameter that indicate signedness of certain operands as this is statically known for each call to the helper functions.
closed 2022-03-10 by mikeurbach