llvm/circt #2750: [Tests] Enable check-circt subtargets

nandor, 2022-03-11:
`ninja check-circt-unit` does not work on its own.

The Unit folder among the tests of CIRCT should expose tests under the `check-circt-unit` target. However, in order to find the `lit.site.cfg.py` files in such a subfolder, `llvm-lit` requires a mapping that is configured specifically for LLVM when generating it from `llvm-lit.in`. The mapping encoded in the LLVM lit is not compatible with CIRCT. The CMake rules in `AddLLVM.cmake` should be modified to pass on this mapping as a command-line argument when `llvm-lit` is used in non-LLVM projects.

dtzSiFive, 2023-07-28:
Unittests work again via linked commit, close?

dtzSiFive, 2023-07-28:
Please re-open if needed, but since this works and runs in CI looks good to close.

closed 2023-07-28 by dtzSiFive