llvm/circt #2759: [SV] Add `sv.for` loop operation

seldridge, 2022-03-12:
Some lowering requires emission of Verilog `for` loops.  E.g., memory randomization from Chisel needs to loop over all entries in the memory and randomize them.  I'm doing this with janked-out `sv.verbatim` in https://github.com/llvm/circt/pull/2757.  It would be much cleaner to be able to have an actual `sv.for` construct to model this.

Note: this may also motivate the need for an `sv.integer` (which is something that Chisel -> Verilog memory randomization currently uses).

mikeurbach, 2022-03-14:
CC @mortbopet because it seems related to https://github.com/llvm/circt/issues/2705. It seems like you guys have come at it from different angles, but are both looking for a way to express a loop in the lower level hardware abstractions.

mortbopet, 2022-03-14:
For my experimentation for #2705, I've been relying on a combination of `scf.for`, a new `sv.generate` operation (to create a SSACFG region inside the graph region of a `hw.module`, and `hw.param.value` to expose module parameters as SSA values to be used in the loop bounds. I've preferred to use `scf` operations since they (as i see it) models identical concepts to SystemVerilog if/for operations.
My goal over there is to be able to do in-CIRCT elaboration of those for-loop structures to expose an elaborated generic module to CIRCT optimization passes. My use-case have considered nesting the loop inside a generate region, but using `scf.for`+`hw.param.value` in `hw.module` scope should also be possible.

At the moment I'm more in favor of using `scf` operations and then having two passes; one which elaborates `scf` operations inside CIRCT and one (i.e., export-verilog) that can emit `scf` operations as their SystemVerilog equivalent.

Off the top of my mind, the only benefit to introducing an explicit `sv.for/scf.if` operation would be the ability to use `hw.module` parameters and parametric expressions directly in the loop bounds/conditional, and i'm not sure that warrants a new operation (especially when `hw.param.value` already exists) - but please correct me if I'm missing something here!

mortbopet, 2022-03-14:
> Put differently, the sv.for that I want is simply "emit this Verilog for loop" which we do not have a good way of doing right now

> That way, any semantic differences between scf and what can be expressed in SV are handled by lowering, like everything else we do in MLIR, and the verilog emitter can simply focus on emitting the correct for loop syntax in SV.

This makes sense to me - separating concerns wrt. SV emission of for loops and the capabilities that we'd like to have inside a 'hw.generate` structure (in-CIRCT elaboration/evaluation) seems like the better approach.

teqdruid, 2023-06-20:
I'm closing this as we have an `sv.for` op and I've seen recent activity around memory initialization using it. Feel free to reopen if I'm mistaken.

closed 2023-06-20 by teqdruid