llvm/circt #2760: A dialect for scheduling problems

jopperm, 2022-03-13:
@mikeurbach and I think it would be useful to have an import/export format for scheduling problem instances, e.g. for writing test cases and benchmarking independent from the concrete (and potentially proprietary!) synthesis flow.

At its core, the problem model in CIRCT consists of a bunch of maps, indexed by MLIR operations and attributes. To that end, it seems appropriate to define a new dialect to represent the problem components *operation*, *operator type*, *dependence* and *instance* as MLIR `Operation`s, and the problem properties attached to the components as MLIR `Attribute`s.

This could look like this: (fully TBD, of course)
```mlir
sched.instance @foo of "CyclicProblem" { initiation_interval=1 } {
  sched.operator_type @cnst { latency=0 }
  sched.operator_type @add { latency=1 }
  sched.operator_type @mul { latency=3 }
  ...
  %0 = sched.operation @op0 deps() auxdeps() { linkedOpr=@cnst, start_time=0 }
  ...
  %4 = sched.operation @op4 deps(%0, %0) auxdeps() { linkedOpr=@add, start_time=3 }
  ...
  %11 = sched.operation @op11 deps(%4) auxdeps (@op17 { distance=2 }) { linkedOpr=@mul, start_time=42 }
  ...
}
```

Going with the dialect approach (instead of just going with Cap'n'Proto, JSON, etc.) allows to generate Python bindings for the scheduling infrastructure with the usual mechanism. This would further benefit prototyping of new problem models and algorithms.

Note: The proposed dialect is intended for testing/prototyping/benchmarking purposes only! Synthesis flows are supposed to use the API provided by the `Problem` class hierarchy directly.

jopperm, 2022-03-13:
Added "good first issue" because this could be an interesting project for someone looking to learn about MLIR dialect implementation -- just 4 ops and a bit of custom parser/printer hacking!

DeepFlyingSky, 2022-03-13:
HI @jopperm  I'm interested in the work with a new dialect. 
And will ask more questions later.

mikeurbach, 2022-03-14:
Awesome, I think this will be a useful addition. Especially excited about python bindings to be able to quickly express new kinds of scheduling problems. I think the sample IR makes a lot of sense as a way to capture the scheduling model (e.g. the representation of deps and auxdeps looks good to me, and storing properties in DictionaryAttrs makes sense).

I agree that this dialect should be designed for testing, etc. rather than "production" flows, and that should be kept clear. To me, it is analogous to circt-opt and all the other MLIR -opt tools: they are a tool for compiler developers, but not really for the end user of a full compiler.

closed 2022-07-29 by jopperm