llvm/circt #2782: [FIRRTL][IMCP] Reg with invalid value for reset and value and propogate the invalid

youngar, 2022-03-17:
This is another attempt to find some bad behaviour in IMCP

```scala
circuit Foo :
  module Foo :
    input clock: Clock
    input p: UInt<1>
    output b: UInt<1>

    inst bar of Bar
    bar.clock <= clock
    bar.a is invalid

    reg r : UInt<1>, clock with: (reset => (p, UInt<1>(1)))
    r <= bar.b
    b <= r

  module Bar :
    input clock : Clock
    input a: UInt<1>
    output b: UInt<1>

    reg r : UInt<1>, clock with: (reset => (a, UInt<1>(1)))
    r is invalid

    b <= r
```
MFC:
```verilog
module Foo(	// ./x.fir:2:10
  input  clock, p,
  output b);

  Bar bar ();	// ./x.fir:7:5
  assign b = 1'h1;	// ./x.fir:2:10, :11:5
endmodule

module Bar();	// ./x.fir:16:10
endmodule
```
SFC:
```verilog
module Foo(
  input   clock,
  input   p,
  output  b
);

  reg  r;
  assign b = r;
  always @(posedge clock) begin
    r <= p;
  end
endmodule
```

seldridge, 2022-03-17:
Awesome find!

Generally, any deviation where IMCP removes a register that SFC does not remove is bad.

As an example, I think we could use this to generate an always failing circuit if we add an assertion that `r` is always zero (i.e., that asserts if the zero interpretation of invalid is not preserved by the compiler):

```scala
node cond = eq(r, UInt<1>(0))
assert(clock, cond, not(reset), "herpderp")
```

This augmentation will work with the SFC and always assert out (after reset) in the MFC.

seldridge, 2022-03-17:
Thinking about this more (and originating from @darthscsi)... I don't think the invalid lattice value is actually useful. Any merge involving an invalid value needs to be treated as a constant zero for SFC exactness. Therefore, there is likely no utility in modeling this lattice value.

A first step would be to merge invalid as if it were zero. A second step would be to cull invalid value from IMCP.

PR #2870 (2022-04-12): [FIRRTL] s/RemoveResets/RemoveInvalid/ and Make All Invalids Zero
This changes the `RemoveResets` pass to be called `RemoveInvalid`.  It then extends this pass to, after invalid reset initializers are converted to reset-less registers, re-walks each module to convert invalids to constant zero ops.  (Invalid bundles are ignored.)  This is done to incorrectly spotty application of the "invalid is zero" interpretation that was applied during canonicalization.

This is done for SFC-exactness, e.g., to fix #2782.
commit cb5d0279 (2022-04-12): [FIRRTL] Squash Invalids in RemoveInvalid

Extend the RemoveInvalid pass (formerly RemoveResets) to convert all
invalid values to zero.  This is done to apply the "invalid is zero"
interpretation after all other invalid interpretations have been
resolved.  This replaces logic where this interpretation was
incompletely applied during canonicalization after RemoveResets.

Fixes #2782.
closed 2022-04-12 by seldridge