llvm/circt #2794: [HW] GlobalRef verifier performance

teqdruid, 2022-03-23:
The internal application on which I'm working contains 1800 global refs. This results in `GlobalRefOp::verify()` being 64% of the runtime. Drilling in further: each time the verifier runs (1800 times per IR verify), it [builds a new `SymbolTable`] (https://github.com/llvm/circt/blob/31205ddd090313e73e2b10462d424c1a56ca9545/lib/Dialect/HW/HWOps.cpp#L1460) which shows as the vast majority of time in `verify()`.

What's the right way to share the `SymbolTable` across all the 1800 `GlobalRefOp::verify()` invocations in a verification pass?

youngar, 2022-03-23:
You can implement `SymbolUserOpInterface` to make this faster.  See FIRRTL's instance op for an example:
https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/FIRRTLOps.cpp#L1294
There is a small issue here where the Op verification order verifies parent ops first and this means that the parent SymbolTable op will call this function before the normal InstanceOp verifiers are run. There is some discussion about this on discourse which I will try to find.

teqdruid, 2022-03-23:
Thanks @youngar! That seemed to work (see my PR).

PR #2796 (2022-03-24): [HW] GlobalRefOp performance enhancement
Instead of building a `SymbolTable` for each verify call, use
`SymbolUserOpInterface` to be fed the symbol table. Fixes #2794.
commit c2936b50 (2022-03-24): [HW] GlobalRefOp performance enhancement (#2796)

Instead of building a `SymbolTable` for each verify call, use
`SymbolUserOpInterface` to be fed the symbol table. Fixes #2794.
closed 2022-03-24 by teqdruid