llvm/circt #2811: [ESI] Services CIRCT model

teqdruid, 2022-03-26:
I've finally started thinking about how [ESI services](https://github.com/llvm/circt/blob/main/docs/ESI/services.md) should look and work in CIRCT, so I'm creating this issue to start the discussion.

Initial proposal for stoning purposes forthcoming. I don't want to bias the discussion by mixing the discussion topic with my initial proposal.

teqdruid, 2022-03-26:
```mlir
// Declare a 'service interface' to be implemented.
esi.service.declaration HostComms single {
  send ToServer : !esi.message
  recieve BroadcastToClients : !esi.channel<i8>
}

// UART implements the service declaration 'HostComms' via a set of RTL-level ports.
// Just like an hw.module.extern, but will have ESI ports implicitly added.
esi.service.implementation @HostComms UART (
    %clk : i1,
    %recv : i8, // Assume we have hard IP which gets bytes.
    %recv_valid : i1 // and it detects when a byte is present.
  ) -> (
    %send : i8, // Assume we have hard IP which sends bytes.
    %send_valid : i1
  )

hw.module @S (...) -> (...) {
  %1 = ... :  !esi.channel<!hw.array<32xi8>>
  // Send to a service anywhere in the design.
  esi.service.add_port @UART::@send %1 randomTextSender : !esi.channel<!hw.array<32xi8>>
  ...
}

hw.module @R (...) -> (...) {
  // Get an ESI port from anywhere in the design.
  %1 = esi.service.get_port @UART::@recieve textParser : !esi.channel<i8>
  ...
}

hw.module @Top (%clk : i1, %recv : i8, %recv_valid : i1) -> (%send : i8, %send_valid : i1) {
  hw.instance r @R (...)
  hw.instance s @S (...)

  // Instantiate a service just like a hw.module.
  %send, %send_valid = esi.service.instance @UART (%clk, %recv, %recv_valid) : i8, i1
  hw.output %send, %send_valid : i8, i1
}
```

Client connections would be implicitly connected to the nearest instance to them when ascending the instance hierarchy. Those connections would, therefore, have to be plumbed through to the module instances. So the above would lower to:

```mlir
hw.module @A (...) -> (..., %1 : !esi.channel<!hw.array<32xi8>>) {
  %1, %bp = esi.wrap %data, %valid : !hw.array<32xi8>
  hw.output ..., %1 : ..., !esi.channel<!hw.array<32xi8>>
}

hw.module @R (..., %1 : !esi.channel<i8>) -> (...) {
  ...
}

// Could also be a hw.module.extern
hw.module @HostComms Top.UART (
    %clk : i1,
    %recv : i8,
    %recv_valid : i1,
    %s_send : !esi.channel<!hw.array<32xi8>>
  ) -> (
    %send : i8,
    %send_valid : i1,
    %r_recv : !esi.channel<i8>
  ) {
  // Generated code.
}

hw.module @Top (%clk : i1, %recv : i8, %recv_valid : i1) -> (%send : i8, %send_valid : i1) {
  hw.instance r @R (..., %r.recv)
  ..., %s.send = hw.instance s @S (...)

  %send, %send_valid, %r.recv = hw.instance @Top.UART (%clk, %recv, %recv_valid, %s.send) : i8, i1
  // Instantiate a service just like a hw.module.
  hw.output %send, %send_valid : i8, i1
}
```

mikeurbach, 2022-03-28:
I'm certainly excited to see this functionality coming. I'll try to dig more into the specific design decisions and collect my thoughts, at least before the next ESI chat.

teqdruid, 2022-04-01:
Based on our discussion this Wednesday morning, I think it is fair to say that there were no major objections to my proposal, right? Most of our discussion was about how best to use services to properly layer concerns (i.e. serialization vs transport).

The biggest open question for me is regarding implementation generators. I think we clearly need to provide several mechanisms (C++ in CIRCT, Python, external tool, etc.). If, however, there are a bunch of different service generators all written in different languages, I suspect composibility would be an issue. Maybe we can get away with some sort of PDL based solution which is intended to instantiate external parameterized Verilog? Or maybe a few standardized C++ components?

makslevental, 2022-04-01:
@teqdruid maybe i'm going to get pilloried for this but what about using [mustache](https://mustache.github.io/mustache.5.html) templates? With bindings available in both Python and C++, anyone could write driver logic. 

Btw it claims it doesn't support conditionals and loops but [apparently it does](https://stackoverflow.com/a/6479017). Not sure if this is a good or bad thing - control flow in the templates is probably a bad thing but maybe you end up needing it (I don't have enough experience with designing generators...).

caveat: I don't have any experience with it but my thinking was "I wonder if there's a templating language with bindings in multiple languages" and mustache was [the only one I found that fit that criterion](https://en.wikipedia.org/wiki/Comparison_of_web_template_engines).

teqdruid, 2022-04-01:
> @teqdruid maybe i'm going to get pilloried for this but what about using [mustache](https://mustache.github.io/mustache.5.html) templates? With bindings available in both Python and C++, anyone could write driver logic.

The question is what would the template be producing? If MLIR IR, then PDL is a better fit IMO. If Python, is a template language really necessary? If SystemVerilog, then I think module parameters are more powerful (and standard) than mustache.

> Also I'm curious to hear @mikeurbach's issues with calling python from CIRCT? I imagine it has something to do with [wrangling pybind in the reverse direction](https://pybind11.readthedocs.io/en/stable/advanced/embedding.html)? I have a bit of experience with this so my thinking is that CIRCT would provide Python drivers for the mustache templates.

Mike and I both have scars from doing this in PyCDE. Originally, PyCDE generators were driven from C++ with the generators themselves written in Python as callbacks. The problem wasn't so much calling Python code from C++, it was the MLIR bindings. They're designed and used for Python API -> IR -> MLIR passes, so once you create the IR you hand it off to some passes and never touch it again. Anything other than that flow has rough edges and deadly thorns.

IIRC the biggest problem we ran into was Operations being erased on the C++ side and the Python side not knowing about it. Any operations on the python objects which had pointers into the IR segfaulted. To make matters worse, when ++ re-uses a pointer for a new op, Python would see it as the old op which doesn't exactly help debugging. Mike may have other scars. Either way, we ended up re-architecting PyCDE to adhere to a write the IR __once__ and don't look back policy.

mortbopet, 2022-04-03:
I've got no objections to the current state of the proposal. As @teqdruid hints to, the biggest question/challenge here will most likely be how we actually get around generating the body of the service during lowering. As for the implementation, I'd like to stick with something that is at least verifiable/typesafe in some way (either MLIR IR (PDL) or a C++ library). So after the ops proposed in this issue lands, it'd be due diligence to try out some of the different styles of generators and find out what feels ergonomic, if it makes sense to abstract this out to a generator interface to be able to hand things off to external tools etc. etc..
A third option (if these generates are _always_ intended to generate RTL IR) could be the `hw.generate` (PR #2828) operation. In any case, however, this would (for now...) only give us the expressiveness of SystemVerilog parametric modules, albeit in-CIRCT.

mikeurbach, 2022-04-04:
> To make matters worse, when ++ re-uses a pointer for a new op, Python would see it as the old op which doesn't exactly help debugging. Mike may have other scars.

Nope, this was the most nefarious issue IIRC.

I will go ahead and say something I've said a lot of times: I've been looking for a project to take PDL for a spin with, could this be it?

That sort of feels like I have a solution and I'm looking for a problem, so another route might be Python bindings. As long as we are generating IR and then handing it off to later passes, I think we can do this in a safe way from Python, which might be the most ergonomic way to start hacking. For a Python example (success story?), I'm thinking about the Linalg OpDsl upstream, which I believe is the blessed way to define Linalg structured ops for some time: https://mlir.llvm.org/docs/Dialects/Linalg/OpDSL/. 

If not, I personally think well layered C++ libraries can go a lot farther than people give credit, and I don't mind writing out the C++ if we take the time to craft a nice library/DSL for our needs here.

Finally, I want to point to `hw.module.generated`, which is more in the camp of "instantiate external parameterized Verilog": https://circt.llvm.org/docs/Dialects/HW/#hwmodulegenerated-circthwhwmodulegeneratedop. As far as I understand, this is essentially a placeholder to stand in until some other tool runs to generate the thing (and there is some support in CIRCT to call the other tool, though I'm fuzzy about this). I'm not sure I'm advocating this approach, but wanted to mention it.

teqdruid, 2022-04-04:
One of the (unstated) goals for services is that board manufacturers could provide their BSPs (Board Support Packages) as a set of ESI services, ideally just implementations of existing, standardized services. So it's important for non-CIRCT developers to be able to construct implementations pretty easily... Which probably points to external, hand coded SystemVerilog as something we'll have to support.

mikeurbach, 2022-04-04:
That makes sense to me. But even though we would want this to support external SystemVerilog, we still want to figure out how to have the compiler generate common implementations, right? I'm kind of looking at this as defining reference implementations that ship with ESI, even if down the road some users are expected to craft their own implementation.

teqdruid, 2022-04-04:
I'm imagining some sort of configuration file which specifies the implementations to use wherein the implementation would be one of a set of C++ 'libraries'. One of which would be instantiating an external parameterized module. Another one would be using the generator call from `hw`. Several would be services implemented in CIRCT directly (like the Cap'nProto serialization code I wrote).

teqdruid, 2022-08-26:
This has been fully implemented!

closed 2022-08-26 by teqdruid