llvm/circt #2820: Errors when building CIRCT

eistar, 2022-03-29:
Hi, I encountered the following errors when build the CIRCT project. It seems that they are about using std::vaiant in Calyx. Could anyone help to check them at your convience?
The log file of ninja build is also enclosed. 
BTW, I have followed the set-up instructions in project mainpage to build it.

Thanks!

```bash
[172/370] Building CXX object lib/Conversion/SCFToCalyx/CMakeFiles/obj.CIRCTSCFToCalyx.dir/SCFToCalyx.cpp.o
FAILED: lib/Conversion/SCFToCalyx/CMakeFiles/obj.CIRCTSCFToCalyx.dir/SCFToCalyx.cpp.o 
/usr/bin/c++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I../llvm/llvm/include -I../llvm/build/include -I../llvm/mlir/include -I../llvm/build/tools/mlir/include -I../include -Iinclude -fno-exceptions -fno-rtti -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wno-comment -Wstring-conversion -fdiagnostics-color -g  -fno-exceptions -fno-rtti -std=gnu++17 -MD -MT lib/Conversion/SCFToCalyx/CMakeFiles/obj.CIRCTSCFToCalyx.dir/SCFToCalyx.cpp.o -MF lib/Conversion/SCFToCalyx/CMakeFiles/obj.CIRCTSCFToCalyx.dir/SCFToCalyx.cpp.o.d -o lib/Conversion/SCFToCalyx/CMakeFiles/obj.CIRCTSCFToCalyx.dir/SCFToCalyx.cpp.o -c ../lib/Conversion/SCFToCalyx/SCFToCalyx.cpp
In file included from ../lib/Conversion/SCFToCalyx/SCFToCalyx.cpp:29:
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:157:13: error: cannot cast private base class 'std::__detail::__variant::_Variant_storage<false, circt::calyx::MemoryOp, circt::CalyxMemoryPorts>' to 'std::variant<circt::calyx::MemoryOp, circt::CalyxMemoryPorts>'
            return static_cast<variant<_Types...>&>(__rhs);
                   ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:408:7: note: in instantiation of function template specialization 'std::__variant_cast<circt::calyx::MemoryOp, circt::CalyxMemoryPorts, std::__detail::__variant::_Variant_storage<false, circt::calyx::MemoryOp, circt::CalyxMemoryPorts> &>' requested here
          }, __variant_cast<_Types...>(*this));
             ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:414:9: note: in instantiation of member function 'std::__detail::__variant::_Variant_storage<false, circt::calyx::MemoryOp, circt::CalyxMemoryPorts>::_M_reset' requested here
      { _M_reset(); }
        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:506:20: note: in instantiation of member function 'std::__detail::__variant::_Variant_storage<false, circt::calyx::MemoryOp, circt::CalyxMemoryPorts>::~_Variant_storage' requested here
      using _Base::_Base;
                   ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:1298:7: note: constrained by private inheritance here
    : private __detail::__variant::_Variant_base<_Types...>,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:157:13: error: cannot cast private base class 'std::__detail::__variant::_Move_assign_base<false, circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>' to 'std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>'
            return static_cast<variant<_Types...>&>(__rhs);
                   ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:666:5: note: in instantiation of function template specialization 'std::__variant_cast<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable, std::__detail::__variant::_Move_assign_base<false, circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> &>' requested here
                  __variant_cast<_Types...>(*this)
                  ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:710:57: note: in instantiation of member function 'std::__detail::__variant::_Move_assign_base<false, circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>::operator=' requested here
      _Variant_base& operator=(_Variant_base&&) = default;
                                                        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:495:22: note: in instantiation of function template specialization 'std::__copy_move<true, false, std::random_access_iterator_tag>::__copy_m<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *, std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *>' requested here
                              _Category>::__copy_m(__first, __last, __result);
                                          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:522:19: note: in instantiation of function template specialization 'std::__copy_move_a2<true, std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *, std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *>' requested here
    { return std::__copy_move_a2<_IsMove>(__first, __last, __result); }
                  ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:530:8: note: in instantiation of function template specialization 'std::__copy_move_a1<true, std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *, std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable> *>' requested here
                std::__copy_move_a1<_IsMove>(std::__niter_base(__first),
                     ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:652:19: note: (skipping 6 contexts in backtrace; use -ftemplate-backtrace-limit=0 to see all)
      return std::__copy_move_a<true>(std::__miter_base(__first),
                  ^
../llvm/llvm/include/llvm/ADT/DenseMap.h:583:13: note: in instantiation of member function 'llvm::DenseMapBase<llvm::DenseMap<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1>, llvm::DenseMapInfo<mlir::Block *, void>, llvm::detail::DenseMapPair<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1> > >, mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1>, llvm::DenseMapInfo<mlir::Block *, void>, llvm::detail::DenseMapPair<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1> > >::grow' requested here
      this->grow(NumBuckets * 2);
            ^
../llvm/llvm/include/llvm/ADT/DenseMap.h:549:17: note: in instantiation of function template specialization 'llvm::DenseMapBase<llvm::DenseMap<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1>, llvm::DenseMapInfo<mlir::Block *, void>, llvm::detail::DenseMapPair<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1> > >, mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1>, llvm::DenseMapInfo<mlir::Block *, void>, llvm::detail::DenseMapPair<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1> > >::InsertIntoBucketImpl<mlir::Block *>' requested here
    TheBucket = InsertIntoBucketImpl(Key, Key, TheBucket);
                ^
../llvm/llvm/include/llvm/ADT/DenseMap.h:328:13: note: in instantiation of function template specialization 'llvm::DenseMapBase<llvm::DenseMap<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1>, llvm::DenseMapInfo<mlir::Block *, void>, llvm::detail::DenseMapPair<mlir::Block *, llvm::SmallVector<std::variant<circt::calyx::GroupOp, circt::WhileScheduleable, circt::PipelineScheduleable>, 1> > >, mlir::Block *, llvm::SmallVector<std: [... cut ...]
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:558:7: note: in instantiation of member function 'std::__detail::__variant::_Copy_ctor_base<false, circt::calyx::MemoryOp, circt::CalyxMemoryPorts>::_Copy_ctor_base' requested here
      _Move_ctor_base(const _Move_ctor_base&) = default;
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/variant:1298:7: note: constrained by private inheritance here
    : private __detail::__variant::_Variant_base<_Types...>,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
11 errors generated.
```
[make.log](https://github.com/llvm/circt/files/8370067/make.log)

mikeurbach, 2022-03-29:
Can you provide some more info on your OS, compiler version, and how you ran CMake?

I'm not 100% sure, but it seems from the logs you are using GCC 11, which should be well supported, but it seems to be including C++11 headers? CIRCT has decided to push forward and adopt C++17, which should be set for the CMake build here: https://github.com/llvm/circt/blob/dbf84306dc9a697cdff766bbbfb03e9ee282916b/CMakeLists.txt#L22-L23.

eistar, 2022-03-30:
Hi Mike,

You pointed me in the right direction.
I found /usr/bin/c++ is pointed to /usr/bin/clang++. However, gcc-11 header files in /usr/include/c++/11 are used during compiling.

After letting /usr/bin/c++ pointed to /usr/bin/g++-11 by update-alternatives, the errors are disappeared.

Thank you so much for the help!

closed 2022-03-30 by eistar

mikeurbach, 2022-03-30:
Awesome, glad you were able to get to the bottom of it.