llvm/circt #2839: [HW] Array*Ops don't accept unpacked arrays

uenoku, 2022-04-03:
Though there is `hw::UnpackedArrayType`, most of operations are not able to handle that type, e.g:

```mlir
 hw.module @Foo(%a:!hw.uarray<3xi1>, %b: i2) -> (e: i1) {
    %c-1_i2 = hw.constant -1 : i2
    %1 = hw.array_get %a[%b] : !hw.uarray<3xi1>
    %2 = comb.icmp eq %b, %c-1_i2 : i2
    %3 = comb.mux %2, %a_0, %1 : i1
    hw.output %3 : i1
  }
```
will cause a verification failure
```mlir
foo.mlir:4:10: error: custom op 'hw.array_get' 'input' must be an ArrayType, but got '!hw.uarray<3xi1>'
```

mikeurbach, 2022-04-04:
I believe this was intentional, but I'm not finding anything about it in the rationale docs. I think this is because unpacked array have trickier semantics in System Verilog, and the things we do with the packed array ops won't always be legal. Sorry for the hand waving. @teqdruid or @darthscsi do either of you have more background on this?

teqdruid, 2022-04-04:
Yeah, `unpacked` is (IMO) a misnomer and it shouldn't be in the `hw` dialect. Hardware doesn't have a notion of packed vs unpacked arrays, so we don't model it in the `hw` dialect. The operations which act on it are in the `sv` dialect -- they're generally have `inout` in the name (e.g. `sv.array_index_inout`, `sv.read_inout`).

I would recommend against `hw.uarray` unless you have a compelling reason -- like something simulation related.

uenoku, 2022-04-04:
@mikeurbach @teqdruid That totally makes sense! Thank you for the explanation. I wanted to try lowering firrtl.multibit_mux into unpacked array to investigate simulation speed difference.

closed 2022-04-04 by uenoku