llvm/circt #2843: [CI] Nightly integration tests

teqdruid, 2022-04-04:
Some of the nightly integration tests have started failing due to not enough disk space on the agents. Mostly gcc debug tests.

Are there some cmake flags we can set to prevent building unrelated files in LLVM/MLIR? (It always bugs me to see OpenMPI and GPU stuff when I compile.) Should we selectively disable the gcc debug tests?

uenoku, 2022-04-04:
I think one thing we can try is to stop building backends other than X86.  I'm not sure how much it helps but I bet it should reduce unrelated lib sizes.

teqdruid, 2022-04-04:
@uenoku good idea!

@mikeurbach The ccache may be the primary source of bloat. The short nightly test uses the ccache action... did not using it on the nightlies make any difference in hit rates?

teqdruid, 2022-04-04:
@mikeurbach Nevermind. The ccaches on failing builds are actually quite small:

[image]

uenoku, 2022-04-05:
Passed CI (https://github.com/llvm/circt/actions/runs/2092757865) with #2845. 
Though I do believe it reaches the limit again in the near future.

teqdruid, 2022-04-05:
> Though I do believe it will reach the limit again in the near future.

That sounds like a problem for future us ;)

Thanks!

commit 060703a0 (2022-04-05): [CI] Only build the host backend (#2845)

Fixes #2843
closed 2022-04-05 by uenoku