llvm/circt #2857: [FIRRTL][InferResets] Incorrect field ID with zero length vectors

youngar, 2022-04-07:
InferResets uses field ids to track specific reset types in aggregates. There is a mismatch in that field ids were not designed to track specific elements types in an aggregate type, but values of an aggregate value.  This is evident in InferRests uses fieldRefs for vectors: it references the first element of the array to represent the element type.  This is problematic when there is a bundle with a 0 length vector, and it ends up targeting the bundle element type after the vector:

```scala
circuit Foo:
  module Foo :
    output out : { resets : Reset[0], flip data : UInt<3> }
    out.resets is invalid
```

```
reset.fir:4:16: error: 'firrtl.strictconnect' op requires all operands to have the same type
    out.resets is invalid
               ^
reset.fir:4:16: note: see current operation: "firrtl.strictconnect"(%0, %1) : (!firrtl.vector<reset, 0>, !firrtl.vector<uint<1>, 0>) -> ()
// -----// IR Dump After InferResets Failed //----- //
"firrtl.circuit"() ({
  "firrtl.module"() ({
  ^bb0(%arg0: !firrtl.bundle<resets: vector<reset, 0>, data flip: uint<1>>):
    %0 = "firrtl.subfield"(%arg0) {fieldIndex = 0 : i32} : (!firrtl.bundle<resets: vector<reset, 0>, data flip: uint<1>>) -> !firrtl.vector<reset, 0>
    %1 = "firrtl.invalidvalue"() : () -> !firrtl.vector<uint<1>, 0>
    "firrtl.strictconnect"(%0, %1) : (!firrtl.vector<reset, 0>, !firrtl.vector<uint<1>, 0>) -> ()
  }) {annotations = [], portAnnotations = [], portDirections = true, portNames = ["out"], portSyms = [], portTypes = [!firrtl.bundle<resets: vector<reset, 0>, data flip: uint<1>>], sym_name = "Foo"} : () -> ()
}) {annotations = [], name = "Foo"} : () -> ()
```

In addition to failing to update the original type, causing a strict-connect verifier error, it changes the `data` field from `uint<3>` to `uint<1>`.

Hopefully this shows the numbering difference for the following bundle type: `{ resets : Reset[0], flip data : UInt<3> }`
Field based numbering:
- 0: `out`
- 1: `out.resets`
- 2: `out.data`

Type based numbering:
- 0: `{ resets : Reset[0], flip data : UInt<3> }`
- 1: `Reset[]`
- 2: `Reset`
- 3: `UInt<3>`

I think the way to solve this is to stop using the generic fieldref utility and reimplement number in the types.

fabianschuiki, 2022-04-07:
I think `InferResets` wants to use field refs for tracking aggregate values, but then confuses a few things and messes up the IDs. Trying to track individual vector element types seems like a bug, since all `InferResets` can do at the end of the day is pick a reset type for _all_ elements at once. The solution here might be to stop resolving individual vector elements 🤔 when finding the sets of connected reset values and wires.

fabianschuiki, 2022-04-07:
Fantastic catch btw @youngar :grin: :tada: !

PR #2858 (2022-04-07): [InferResets] Fix inference for zero-length vectors
When tracing resets through the design, the `InferResets` pass always uses the ID of the first vector element if it encounters a vector type. This is done to ensure that connections to an element of a vector affects the type inference of all the other elements too, since the pass can only pick a single reset type for the entire vector instead of individual types for each vector element.

This is problematic since it assumes that all vectors have at least one element, which may not be the case in practice. If such a vector is encountered, the  [... cut]
commit 9a168eed (2022-04-07): [InferResets] Fix inference for zero-length vectors (#2858)

When tracing resets through the design, the `InferResets` pass always
uses the ID of the first vector element if it encounters a vector type.
This is done to ensure that connections to an element of a vector
affects the type inference of all the other elements too, since the pass
can only pick a single reset type for the entire vector instead of
individual types for each vector element.

This is problematic since it assumes that all vectors have at least one
element, which may not be the case in practice. If such a vector is
encounte [... cut]
closed 2022-04-07 by fabianschuiki