llvm/circt #2863: [ExportVerilog] Named constants changes how the `not` operation is printed

youngar, 2022-04-08:
```
circuit Foo:
  module Foo:
    input i: UInt<1>
    output o : UInt<1>
    wire w: UInt<1>
    w <= UInt<1>(1)
    o <= not(i)
```
I would have liked to see `assign o = ~i` in the output:
```
module Foo(	// ./names.fir:2:10
  input  i,
  output o);

  wire w;	// ./names.fir:5:5

  assign w = 1'h1;	// ./names.fir:5:5, :6:10
  assign o = i ^ w;	// ./names.fir:2:10, :5:5, :7:10
endmodule
```

uenoku, 2022-04-08:
I guess this should be fixed by https://github.com/llvm/circt/pull/2851 (wire will not be created for a constant) 
https://github.com/llvm/circt/blob/0726a647e694a65a83481784719c22910c08c50a/lib/Dialect/SV/Transforms/PrettifyVerilog.cpp#L264-L267

youngar, 2023-04-26:
This is currently printing the correct thing:
```verilog
module Foo(
  input  i,
  output o
);

  assign o = ~i;
endmodule
```

closed 2023-04-26 by youngar