llvm/circt #2888: [FIRRTL] Don't Dedup Modules with Different NestedPrefixModulesAnnotation

seldridge, 2022-04-13:
`NestedPrefixModulesAnnotation` with different prefixes should not deduplicate. This can be interpreted one of two ways, either of which should have the same effect:

1. `NestedPrefixModulesAnnotation` should block deduplication.
2. The `PrefixModules` pass should duplicate modules which have different prefixes.

SiFive's extensions to the SFC are implementing this using strategy (2).

An end-to-end test case that should work is the following:

```scala
circuit Foo: %[[
  {
    "class":"sifive.enterprise.firrtl.NestedPrefixModulesAnnotation",
    "prefix":"prefix1_",
    "inclusive":true,
    "target":"~Foo|Bar"
  },
  {
    "class":"sifive.enterprise.firrtl.NestedPrefixModulesAnnotation",
    "prefix":"prefix2_",
    "inclusive":true,
    "target":"~Foo|Baz"
  }
]]
  module Bar:
  module Baz:
  module Foo:
    inst bar of Bar
    inst baz of Baz
```

Currently, this produces (`firtool -dedup -strip-debug-info`):

```verilog
module prefix2_Bar();
endmodule

module Foo();
  prefix2_Bar bar ();
  prefix2_Bar baz ();
endmodule
```

This should produce:
```verilog
module prefix1_Bar();
endmodule

module prefix2_Baz();
endmodule

module Foo();
  prefix1_Bar bar ();
  prefix2_Baz baz ();
endmodule
```

There may also be an interaction here with memories where memories should not deduplicate if they have different prefixes.  This may be a little funky because it has interactions with the memory generator schema which may not respect prefixing.

youngar, 2022-04-13:
I kind of think we need to implement both options 1 and 2. The `NestedPrefixModulesAnnotation` is currently "sticky" similar to the `InlineModule` annotation, and fixing Option 2 means fixing this.  This is probably fairly obvious, but this means that the `PrefixModules` pass treats non-local annotations as local:
```mlir
firrtl.circuit "Foo"  {
  firrtl.nla @nla_0 [#hw.innerNameRef<@Foo::@bar>, @Bar]
  firrtl.nla @nla [#hw.innerNameRef<@Foo::@baz>, @Bar]
  firrtl.module private @Bar() attributes {annotations = [{circt.nonlocal = @nla, class = "sifive.enterprise.firrtl.NestedPrefixModulesAnnotation", inclusive = true, prefix = "prefix2_"}, {circt.nonlocal = @nla_0, class = "sifive.enterprise.firrtl.NestedPrefixModulesAnnotation", inclusive = true, prefix = "prefix1_"}]} {
  }
  firrtl.module @Foo() {
    firrtl.instance bar sym @bar  {annotations = [{circt.nonlocal = @nla_0, class = "circt.nonlocal"}]} @Bar()
    firrtl.instance baz sym @baz  {annotations = [{circt.nonlocal = @nla, class = "circt.nonlocal"}]} @Bar()
  }
}

// -----// IR Dump After PrefixModules //----- //
firrtl.circuit "Foo"  {
  firrtl.nla @nla_0 [#hw.innerNameRef<@Foo::@bar>, @prefix2_Bar]
  firrtl.nla @nla [#hw.innerNameRef<@Foo::@baz>, @prefix2_Bar]
  firrtl.module private @prefix2_Bar() {
  }
  firrtl.module @Foo() {
    firrtl.instance bar sym @bar  {annotations = [{circt.nonlocal = @nla_0, class = "circt.nonlocal"}]} @prefix2_Bar()
    firrtl.instance baz sym @baz  {annotations = [{circt.nonlocal = @nla, class = "circt.nonlocal"}]} @prefix2_Bar()
  }
}
```

> There may also be an interaction here with memories where memories should not deduplicate if they have different prefixes. This may be a little funky because it has interactions with the memory generator schema which may not respect prefixing.

I think we will need some separate mechanism to prevent the deduplication of memories, and maybe we should start thinking about how we will implement "deduplication domains". Memories are deduplicated as a part of [LowerToHw](https://github.com/llvm/circt/blob/main/lib/Conversion/FIRRTLToHW/LowerToHW.cpp#L604) (I think? @prithayan is the expert here).

seldridge, 2022-04-13:
Here's the equivalent, verbose example of the above, but using memories:

```scala
circuit Foo : %[[
  {
    "class":"sifive.enterprise.firrtl.NestedPrefixModulesAnnotation",
    "prefix":"prefix1_",
    "inclusive":true,
    "target":"~Foo|Bar"
  },
  {
    "class":"sifive.enterprise.firrtl.NestedPrefixModulesAnnotation",
    "prefix":"prefix2_",
    "inclusive":true,
    "target":"~Foo|Baz"
  }
]]
  module Bar :
    input clock : Clock
    input reset : Reset
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    mem mem :
      data-type => UInt<1>
      depth => 8
      read-latency => 1
      write-latency => 1
      reader => readData_MPORT
      writer => MPORT
      read-under-write => undefined
    mem.readData_MPORT.addr is invalid
    mem.readData_MPORT.clk is invalid
    mem.readData_MPORT.en <= UInt<1>("h0")
    mem.MPORT.addr is invalid
    mem.MPORT.clk is invalid
    mem.MPORT.en <= UInt<1>("h0")
    mem.MPORT.data is invalid
    mem.MPORT.mask is invalid
    mem.readData_MPORT.addr <= readAddr
    mem.readData_MPORT.clk <= clock
    readData <= mem.readData_MPORT.data
    when writeEn :
      mem.MPORT.addr <= writeAddr
      mem.MPORT.clk <= clock
      mem.MPORT.en <= UInt<1>("h1")
      mem.MPORT.mask <= UInt<1>("h0")
      mem.MPORT.data <= writeData
      mem.MPORT.mask <= UInt<1>("h1")

  module Baz :
    input clock : Clock
    input reset : Reset
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    mem mem :
      data-type => UInt<1>
      depth => 8
      read-latency => 1
      write-latency => 1
      reader => readData_MPORT
      writer => MPORT
      read-under-write => undefined
    mem.readData_MPORT.addr is invalid
    mem.readData_MPORT.clk is invalid
    mem.readData_MPORT.en <= UInt<1>("h0")
    mem.MPORT.addr is invalid
    mem.MPORT.clk is invalid
    mem.MPORT.en <= UInt<1>("h0")
    mem.MPORT.data is invalid
    mem.MPORT.mask is invalid
    mem.readData_MPORT.addr <= readAddr
    mem.readData_MPORT.clk <= clock
    readData <= mem.readData_MPORT.data
    when writeEn :
      mem.MPORT.addr <= writeAddr
      mem.MPORT.clk <= clock
      mem.MPORT.en <= UInt<1>("h1")
      mem.MPORT.mask <= UInt<1>("h0")
      mem.MPORT.data <= writeData
      mem.MPORT.mask <= UInt<1>("h1")

  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    inst bar of Bar
    bar.clock <= clock
    bar.reset <= reset
    bar.readAddr <= readAddr
    bar.writeEn <= writeEn
    bar.writeAddr <= writeAddr
    bar.writeData <= writeData
    inst baz of Baz
    baz.clock <= clock
    baz.reset <= reset
    baz.readAddr <= readAddr
    baz.writeEn <= writeEn
    baz.writeAddr <= writeAddr
    baz.writeData <= writeData
    node _readData_T = xor(bar.readData, baz.readData)
    readData <= _readData_T
```

Eventually, the SiFive SFC gets this into the following form, noting that the memories are duplicated:
```scala
circuit Foo :
  module prefix1_Bar :
    input clock : Clock
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    inst mem of prefix1_mem
    readData <= pad(mem.R0_data, 32)
    mem.R0_addr <= readAddr
    mem.R0_clk <= clock
    mem.W0_addr <= writeAddr
    mem.W0_en <= writeEn
    mem.W0_clk <= clock
    mem.W0_data <= bits(writeData, 0, 0)

  module prefix2_Bar :
    input clock : Clock
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    inst mem of prefix2_mem
    readData <= pad(mem.R0_data, 32)
    mem.R0_addr <= readAddr
    mem.R0_clk <= clock
    mem.W0_addr <= writeAddr
    mem.W0_en <= writeEn
    mem.W0_clk <= clock
    mem.W0_data <= bits(writeData, 0, 0)

  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input readAddr : UInt<3>
    output readData : UInt<32>
    input writeEn : UInt<1>
    input writeAddr : UInt<3>
    input writeData : UInt<32>

    inst bar of prefix1_Bar
    inst baz of prefix2_Bar
    readData <= xor(bar.readData, baz.readData)
    bar.clock <= clock
    bar.readAddr <= readAddr
    bar.writeEn <= writeEn
    bar.writeAddr <= writeAddr
    bar.writeData <= writeData
    baz.clock <= clock
    baz.readAddr <= readAddr
    baz.writeEn <= writeEn
    baz.writeAddr <= writeAddr
    baz.writeData <= writeData

  extmodule mem_ext :
    input R0_addr : UInt<3>
    input R0_en : UInt<1>
    input R0_clk : Clock
    output R0_data : UInt<1>
    input W0_addr : UInt<3>
    input W0_en : UInt<1>
    input W0_clk : Clock
    input W0_data : UInt<1>
    defname = mem_ext

  module prefix1_mem :
    input R0_addr : UInt<3>
    input R0_clk : Clock
    output R0_data : UInt<1>
    input W0_addr : UInt<3>
    input W0_en : UInt<1>
    input W0_clk : Clock
    input W0_data : UInt<1>

    inst mem_ext of mem_ext
    mem_ext.R0_clk <= R0_clk
    mem_ext.R0_en <= UInt<1>("h0")
    mem_ext.R0_addr <= R0_addr
    R0_data <= mem_ext.R0_data
    mem_ext.W0_clk <= W0_clk
    mem_ext.W0_en <= W0_en
    mem_ext.W0_addr <= W0_addr
    mem_ext.W0_data <= W0_data

  module prefix2_mem :
    input R0_addr : UInt<3>
    input R0_clk : Clock
    output R0_data : UInt<1>
    input W0_addr : UInt<3>
    input W0_en : UInt<1>
    input W0_clk : Clock
    input W0_data : UInt<1>

    inst mem_ext of mem_ext
    mem_ext.R0_clk <= R0_clk
    mem_ext.R0_en <= UInt<1>("h0")
    mem_ext.R0_addr <= R0_addr
    R0_data <= mem_ext.R0_data
    mem_ext.W0_clk <= W0_clk
    mem_ext.W0_en <= W0_en
    mem_ext.W0_addr <= W0_addr
    mem_ext.W0_data <= W0_data
```

seldridge, 2022-04-13:
> The NestedPrefixModulesAnnotation is currently "sticky" similar to the InlineModule annotation, and fixing Option 2 means fixing this. 

I actually don't think it is sticky? For the module example, after dedup, the circuit is:

```mlir
firrtl.circuit "Foo"  {
  firrtl.nla @nla_0 [#hw.innerNameRef<@Foo::@bar>, @Bar]
  firrtl.nla @nla [#hw.innerNameRef<@Foo::@baz>, @Bar]
  firrtl.module private @Bar() attributes {annotations = [{circt.nonlocal = @nla, class = "sifive.enterprise.firrtl.NestedPrefixModulesAnnotation", inclusive = true, prefix = "prefix2_"}, {circt.nonlocal = @nla_0, class = "sifive.enterprise.firrtl.NestedPrefixModulesAnnotation", inclusive = true, prefix = "prefix1_"}]} {
  }
  firrtl.module @Foo() {
    firrtl.instance bar sym @bar  {annotations = [{circt.nonlocal = @nla_0, class = "circt.nonlocal"}]} @Bar()
    firrtl.instance baz sym @baz  {annotations = [{circt.nonlocal = @nla, class = "circt.nonlocal"}]} @Bar()
  }
}
```

The prefix annotations are promoted to NLAs, i.e., no information is destroyed. It sounds more like module prefixing needs to know what to do with NLAs or it is applying the module prefixing using a "sticky" interpretation. 🤔

youngar, 2022-04-13:
> I actually don't think it is sticky? 
Yeah I guess I wasn't being super clear: MFC's dedup doesn't have special handling for "sticky" annotations: they are always converted to non-local.  What I meant is that the `PrefixModules` pass treats the annotation as local ("stick interpretation").