llvm/circt #2892: [Calyx] Compiling structural Calyx to `hw`

rachitnigam, 2022-04-13:
Currently, we can roundtrip Calyx programs between the [native][] infrastructure and the CIRCT dialect for Calyx. This means that we can, in essence, run Calyx optimizations in both the systems and then have the native compiler [lowers Calyx programs][lowering] into a group and control-free representation. This representation is intentionally designed to be very close to RTL programs and the native compiler transliterates assignments in Calyx to mutliplexed assignments in SystemVerilog.

We should have a similar lowering pass in the CIRCT infrastructure that can compile control-free Calyx programs into the `hw` dialect. This will unlock the optimizations performed by the `comb` and `seq` compilers and allow us to build a robust RTL backend for Calyx.

With this in hand, once the Calyx lowering passes are reimplemented in CIRCT (https://github.com/llvm/circt/issues/2375), we can have a standalone Calyx-CIRCT infrastructure that is independent of the native infrastructure. The [lowering][] pass in Calyx is a good starting point for reimplementing this in CIRCT.

Additionally, do people have opinions on how to use the `comb`, `seq`, and `hw` dialects for this? At a high-level, we need access to verilog's continuous assignment statements (`assign`) and ternary expressions. We can probably use the `comb` primitives to realize the guard expressions for the ternaries. We also need to use `module`s to talk about component instantiation in Calyx.

[native]: https://github.com/cucapra/calyx
[lowering]: https://github.com/cucapra/calyx/blob/master/src/backend/verilog.rs

cgyurgyik, 2022-04-16:
I think we had planned to do this earlier, and it just got lost in the mix.

https://github.com/llvm/circt/blob/main/lib/Conversion/CalyxToHW/CalyxToHW.cpp

rachitnigam, 2022-04-18:
Sketching out the design a bit more, the majority of the lowering process generates SystemVerilog assignments that look like this:
```
assign port = guard_1 ? out_1 : 
                       guard_2 ? out_2 : 
                       ...
                       guard_n ? out_n : out_{n+1}
```

The ternary expressions can be encoded using [comb.mux](https://circt.llvm.org/docs/Dialects/Comb/#combmux-circtcombmuxop) and the assignments using [sv.assign](https://circt.llvm.org/docs/Dialects/SV/#svassign-circtsvassignop). Obviously, this `comb.mux` stuff will generate priority logic for the guards which can be slow so we should move to `case` based encoding in the future. 

We can also use [comb.parity](https://circt.llvm.org/docs/Dialects/Comb/#combparity-circtcombparityop) and [sv.assert](https://circt.llvm.org/docs/Dialects/SV/#svassert-circtsvassertop) to generate statements to check if more than one guard was true during simulation. These kinds of guards are generated by the Verilator backend for the native compiler and have been pretty helpful.

The remainder of the encoding can probably use things like [hw.module](https://circt.llvm.org/docs/Dialects/HW/#hwmodule-circthwhwmoduleop) and [hw.instance](https://circt.llvm.org/docs/Dialects/HW/#hwinstance-circthwinstanceop) to talk about Calyx components and sub-components.

mikeurbach, 2022-04-18:
Just a small thing: it'd be nice to avoid SV dialect, except for the cases we want the verification statements like assert. I think it should be possible to express the conditional logic and dataflow with comb.mux and hw, without having to drop to SV wires/assigns. Staying firmly in the Comb dialect should admit more optimizations opportunities, and ExportVerilog should take care of emitting the necessary `assign` statements in the end.

mikeurbach, 2022-04-19:
Well, in Comb, there is no concept of wires that can be assigned to, there are just SSA values flowing through a graph. To "assign" an SSA value in the HW dialect, it can be passed as an operand to `hw.instance` or `hw.output`, which model structural connections. In your original example, you had `assign port ...`, so I assume you were indicating assigning to an output, but if you have a different use-case maybe I can help.

rachitnigam, 2022-04-19:
Ah, generally it would be some wire connected to the input of another instance (sub-component) that is being used by the module. Something like:
```
std_add #(32) a0 (.left(a0_left), ...)

assign a0_left = g1 ? r0.out : r1.out;
```

mikeurbach, 2022-04-19:
so in comb/hw that would look something like (psuedo-IR):

```mlir
%0 = hw.instance @std_add a0(%left, ...)
%left = comb.mux %g1, %r0_out, %r1_out
```

The result of comb.mux becomes an operand of hw.instance. Not that ExportVerilog may or may not emit a wire in this case, depending on a fairly complex set of conditions.

Also, I'm pretty sure we would be modeling std_add with a comb.add anyway, so it'd look more like:

```mlir
%0 = comb.add %left, ...
%left = comb.mux %g1, %r0_out, %r1_out
```

(note that these ops will exist in the body of an hw.module, which doesn't enforce def-before-use, so the above ordering is valid. in all likelihood, the ir would often be emitted in a form where defs proceed uses).

cgyurgyik, 2022-04-20:
+1 to lowering to HW operations, since then we don't need to reinvent things such as constant folding for the Calyx dialect.

mikeurbach, 2022-05-16:
The initial version of this is in, and modulo some primitives that I just didn't add conversions for, this should support most Calyx IR; in other words, the guts are all there, and we can have small follow ups to add further support.

One thing I changed my mind about since this issue was opened, the implementation makes extensive use of wires and assigns in the SV dialect. This makes bridging the Calyx world to the CIRCT core world relatively straightforward in a generic way. And in many cases, it leads to the IR I outlined above that is just a combinational dataflow graph, _after_ canonicalization removes unnecessary wires. This seemed like a win once I realized single writer, single reader wires are already canonicalized away.

I'd be in favor of closing this and opening new issues for any missing support of lowering errors, does that sound good @rachitnigam ?

rachitnigam, 2022-05-16:
Yup, sounds good! As I mentioned in the other PR. A reasonable next goal is to lower all the correctness tests in the Calyx test suite (https://github.com/cucapra/calyx/tree/master/tests/correctness) and see if they retain functionality. An easy way to do this would be to add a new `fud` stage that invoke the MLIR stack to compile Calyx programs to verilog.

mikeurbach, 2022-05-16:
Sounds good, that would be interesting to hook up in `fud`. I will close this and we can open an issue over there for such an integration.

closed 2022-05-16 by mikeurbach