llvm/circt #2898: [CI] Nightly integration tests broken

teqdruid, 2022-04-14:
GCC builds are broken https://github.com/llvm/circt/runs/6023506634?check_suite_focus=true . Looks like some upstream code broke them.

@richardxia I think this was your recent LLVM submodule bump: https://github.com/llvm/circt/commit/654d5e96c84c80274f04ddd1a5984adc547b5cec

richardxia, 2022-04-14:
Looks like this is the LLVM bump that will never cease to have issues! Thanks for paying attention to those, I'll start looking at it now.

teqdruid, 2022-04-14:
Since I'm the one who first set up the pipeline, I'm the only one who gets
email notifications when it fails.

richardxia, 2022-04-14:
I found this post on Discourse from a week and a half ago saying that the minimum toolchain versions will be bumped: https://discourse.llvm.org/t/important-new-toolchain-requirements-to-build-llvm-will-most-likely-be-landing-within-a-week-prepare-your-buildbots/61447

That bump landed in https://reviews.llvm.org/D122976 (https://github.com/llvm/llvm-project/commit/4c72deb613d9d8838785b431facb3eb480fb2f51).

It looks like it was only supposed to increase the minimum GCC version to 7, but I think we're already running that version of GCC in the nightly tests. I see that we're running [Ubuntu 18.04](https://github.com/llvm/circt/blob/34277cacf11dc50346f4829a2747e4c94596519f/.github/workflows/nightlyIntegrationTests.yml#L14), which [should be using GCC 7.4](https://packages.ubuntu.com/bionic/gcc).

I'll continue digging into it.

richardxia, 2022-04-14:
Sorry, I think everything that I brought up just now is a red herring. All that patch did was bump the minimum tool versions in the CMake files, causing them to raise soft errors, but that's clearly not the issue we're running into here.

richardxia, 2022-04-14:
I got a bunch of help from @youngar to investigate this, and he found that the error in question was fixed upstream shortly afterwards, in https://github.com/llvm/llvm-project/commit/932f27dc1f0334eb60e06d34536d93674c923672.

Unfortunately, there are still other issues I've been hitting while trying to get up to date with the latest upstream LLVM, which I'm still working through, but I think if we can get that bump in, then the nightly tests should be fixed.

teqdruid, 2022-04-14:
Sounds good!

The other configurations in the nightly pipeline get cancelled when one configuration fails. @mikeurbach / @youngar is there an option we can set to avoid this?

richardxia, 2022-04-14:
I think if we set this [`fail-fast`](https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#jobsjob_idstrategyfail-fast) setting to `false`, it'll allow the rest of the jobs in the matrix to continue running.

mikeurbach, 2022-04-14:
Yep was about to send this https://docs.github.com/en/github-ae@latest/actions/using-workflows/workflow-syntax-for-github-actions#jobsjob_idstrategyfail-fast.

I can send a patch.

mikeurbach, 2022-04-14:
PR: https://github.com/llvm/circt/pull/2899
Nightly test run manually triggered on this branch to see if it works: https://github.com/llvm/circt/actions/runs/2170061034

richardxia, 2022-04-15:
https://github.com/llvm/circt/pull/2901 will bump us to the latest LLVM, which should hopefully fix the GCC compilation issue.

I kicked off a build at https://github.com/llvm/circt/runs/6033987037?check_suite_focus=true

richardxia, 2022-04-16:
Looks like the [latest nightly build](https://github.com/llvm/circt/actions/runs/2176698841) succeeded, so I'm going to close this ticket.

closed 2022-04-16 by richardxia

teqdruid, 2022-04-16:
Thanks again!

richardxia, 2022-04-16:
No problem, and thanks for staying on top of these!