llvm/circt #2909: Helper for llvm::Log2_64_Ceil or 1

mikeurbach, 2022-04-16:
Since we don't support 0 width wires (https://github.com/llvm/circt/issues/2219), there are a lot of situations where we take the size of something (usually a type), and if it is 1, return 1, else return Log2_64_Ceil. This is common in situations like "How many bits do I need to index into this array?". Because we can't say zero bits for a 1 element array (yet), we use 1 bit. This logic gets sprinkled throughout the codebase (sometimes with alternate styles), to the point that I think it's worth having a helper, potentially in lib/Support, to do this so we have a consistent way of doing it. Good first issue, since it's simple to implement, and hunting down instances of Log2_64_Ceil is a good tour of CIRCT.

teqdruid, 2022-04-16:
I think we should just do the `i0` enablement instead.

mikeurbach, 2022-04-16:
Either way, I'd just like to clean up all the places we have this pattern in the codebase :) If it's not tricky to support 0 width wires any more, that would be great.

cgyurgyik, 2022-04-17:
SCFToCalyx case:

https://github.com/llvm/circt/blob/7ce00e745e8af7aa78013486413ae0d76fe5ac9a/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L1094-L1098