llvm/circt #2923: Add `calyx.primitive` operator

rachitnigam, 2022-04-19:
The Calyx `primitive` operator is used to describe properties of verilog externs that the compiler links the final design to. This would be a useful operator to add into the Calyx dialect so that we can start emitting full programs with externals precisely described in CIRCT.

There are two parts to this: (1) The `extern` region that points to a string path which contains all the Verilog definition and contains the shims for all the `primitive` definition in it, and (2) the `primitive` operator itself which is very similar to a `component` except that it allows for parameters. I think it can be very similar to `hw.module` style parameterization.

For example, see the [core primitives file](https://github.com/cucapra/calyx/blob/master/primitives/core.futil) in Calyx.

mikeurbach, 2022-04-19:
One thing I would like to try to shoot for here is to be able to define primitives in terms of "core" CIRCT dialects, not just hand written Verilog. We could start with a `calyx.primitive` that is basically `hw.module.extern` (i.e. placeholder for Verilog you bring later), but I think it would be really powerful to extend that and replace opaque Verilog with IR that CIRCT can reason about. 

I guess what I'm saying is perhaps we don't just follow what the Rust compiler did here, and think about how Calyx can benefit from tying into the broader CIRCT ecosystem in a first class way. For example, what does it look like to write your primitives in Chisel, compile them with FIRRTL, and link them to Calyx via the `primitive` mechanism?

rachitnigam, 2022-04-19:
Yup, I think that’s a good idea. We’ll immediately see benefits if we describe things like std_add as comb.add which will let the CIRCT comb optimization do their thing

mikeurbach, 2022-04-19:
That is one thing, but I'm also interested in new primitives that never existed in a Calyx primitive library before. For example: there's a systolic array op: https://circt.llvm.org/docs/Dialects/MSFT/#msftsystolicarray-circtmsftsystolicarrayop. It would be cool to be able to plug that in as a primitive. To restate: I'm interested in making all well-defined CIRCT dialect ops potential primitives, but I'm also interested in designers being able to define new primitives and Calyx working with them (which is more what the previous comment was about).

mikeurbach, 2022-04-19:
> If neither of those happen, then they can only be co-optimized once Calyx is lowered all the way to hw itself.

Sorry, yes that is spot on, and that is what I'm shooting for. I just wanted to be clear that I don't think the primitives should be implemented in Verilog. In the initial comment, there was this part about "(1) The extern region that points to a string path which contains all the Verilog definition", and I wanted to bring up that CIRCT should target primitives in the HW/Comb/etc. dialects, so once Calyx IR is lowered to those dialects, everything can be stitched together and co-optimized. I just wanted to make this explicit so if someone comes along to work on this, they don't cargo cult the Rust implementation where primitives are Verilog files.

That said, I think there are some interesting long term things we can do here:

> (1) Calyx passes understand what they are

This is actually not that bad with MLIR, specifically using op interfaces. For example, check out the BranchOpInterface or RegionBranchOpInterface. These allow compilers to generically reason about control flow without knowing anything about the ops implementing the interface. If the Calyx dialect defines the interfaces, and people bringing their own primitives implement them, we could write generic passes without Calyx needing to understand the specific primitives.

> (2) they get lowered into Calyx programs themselves

This is also an option, and we could definitely do it for the systolic arrays, for example.

cgyurgyik, 2022-04-20:
Agree with Mike. This sounds like a "how Calyx primitives should be lowered" problem. More for context, we currently support Calyx primitives as shown in [circt/include/circt/Dialect/Calyx/CalyxPrimitives.td](https://github.com/llvm/circt/blob/958a3bbe786b6116036c89d9bebe22e7d0c6e70d/include/circt/Dialect/Calyx/CalyxPrimitives.td) 

Each primitive is a sub-class of CalyxPrimitive, which itself is a sub-class of CalyxCell. Parameterized types are part of the builder, e.g., 

```mlir
// 32-bit register
 %r.in, %r.write_en, %r.clk, %r.reset, %r.out, %r.done = calyx.register @r : i32, i1, i1, i1, i32, i1
```

Perhaps a better approach is to turn CalyxPrimitive into some type of `OpInterface`, that would provide the Calyx compiler enough information to do its job, such as cycle count for static analyses or read/write hazards.

cgyurgyik, 2022-05-27:
I'm not sure I understand the goal of this. What benefit does this have over the current method, i.e., let conversion passes describe how a Calyx primitive is interpreted in another dialect? It seems pretty brittle to tie Calyx primitives to some hard-coded implementation.

rachitnigam, 2022-05-27:
Not all primitives can de defined in terms of conversions into other primitives in a dialect. For example, if I want to interface with a predefined verilog IP, I don't really have any way of compiling it into another dialect. Calyx's `primitive` declarations, much like externs in other languages let's us get around this problem by defining the interface to something without describing exactly what code implements it.

cgyurgyik, 2022-05-27:
Let me summarize my understanding: the goal here is to add an externally defined body to a Calyx primitive for some dialect X. Then, when it is lowered to X, it defaults to using that body? I still see this as being part of the lowering conversion to dialect X. Let Calyx primitives define an interface, and lowering conversions provide the body.

There could be some argument made for custom ops, but I don't really see that even being the case in Calyx native either.

>  if I want to interface with a predefined verilog IP, I don't really have any way of compiling it into another dialect

What do you mean? Why are you compiling a predefined verilog IP in Calyx?

mikeurbach, 2022-05-27:
> Why are you compiling a predefined verilog IP in the Calyx dialect?

Every real-world use case of a hardware compiler I've seen has had a need for something like this. Maybe it's an ASIC memory that a memory compiler will generate or a DSP on a specific Xilinx FPGA board. This is the same reason `hw.module.extern` exists in the HW dialect.

Let me try to back up and try to re-phrase why we are doing this. I actually put my thoughts down earlier over here: https://github.com/cucapra/calyx/discussions/969.

In short, the CalyxPrimitive that currently exists in tablegen is used to define a bunch of ops, which implement the CalyxCell interface, and more or less map to the full library of primitives the native Calyx compiler defines. But what I would like is a way to extend Calyx with new primitives ops, that work with the Calyx infrastructure just like all the existing ones. A good way to do this is with something like `hw.module.extern` that represents a declaration, but not a definition (to borrow C-like language terminology). In other words, an op that declares an interface, but says nothing about an implementation. Calyx passes should be able to reason about such declarations (i.e. the op should probably implement the CalyxCell interface). We can use them in Groups, etc.

If I understand your feedback, you're suggesting that this is inherently lowering defined anyway, so why bother having the new declaration op at all? In other words, if I want to stamp out a bunch of primitives for Xilinx DSPs, I could add a bunch of ops that inherit CalyxPrimitive in tablegen, and define appropriate conversions in a pass like how lower-to-hw works. Is that fair? I would say that's also possible, but less extensible. It requires declaring new ops that depend on the Calyx dialect, whereas if an op like `calyx.primitive` exists, new flows can target it, and lower from it, without declaring any new ops in tablegen.

Another way of putting it: `calyx.primitive` would allow Calyx to keep maintaining a "closed world" stance about not mixing with other dialects, but still maintain a level of extensibility. In fact, `calyx.primitive` could probably subsume all of the uses of CalyxPrimitive as they are today. Instead of a bunch of ops inheriting `CalyxPrimitive` like this: https://github.com/llvm/circt/blob/3bc2d4762cba98463581d77b547a673dc6758df5/test/Dialect/Calyx/emit.mlir#L23
we'd just have one op: `calyx.primitive`, with many different instantiations, like `calyx.primitive "std_mult_pipe" @static(3) : i1, i1, i1, i32, i32, i32, i1`. Note that I am giving a hypothetical syntax with `@static(3)`, but that is a hint at the power of an op like this: with one construct we could capture the entire standard library, including details like static timing as well as port types in a well-defined and generic way. Having one op rather than many makes the IR "[smoother](https://dl.acm.org/doi/abs/10.1145/1115858.1115870)".

mikeurbach, 2022-05-27:
I haven't looked at @rachitnigam's PR at all, but I will try to take a look over the weekend. Thanks for getting the ball rolling on this!

rachitnigam, 2022-05-27:
Yup! Mike hits the nail on its head---you always have to interact with the outside world. Another thing to consider is that "core" primitives can remain as `CalyxPrimitive` if that makes it easier to compile them into `comb` and `seq` dialects but everything else can be a `calyx.primitive`.

To add a final point, `calyx.primitive` also makes it easier to experiment with new primitives–I don't have to recompile the compiler to try using some new Verilog

mikeurbach, 2022-05-27:
> Another thing to consider is that "core" primitives can remain as CalyxPrimitive if that makes it easier to compile them into comb and seq dialects but everything else can be a calyx.primitive.

Probably makes sense for the short term, since we finally got the CalyxToHW pass working with these 😅. I guess what I should say is for the long term, it'd be a measure of success IMO if the "core" primitives were no more privileged than others.

cgyurgyik, 2022-05-27:
Ok thanks that makes more sense. So what we're trying to do is something like:

```mlir
// A program that targets SystemVerilog.
calyx.program {

calyx.primitive_definition @my_verilog_fun ... {
  // some SV-dialect shenanigans
}

calyx.component @C () -> () {
  %0, %1, %2 = calyx.primitive @my_verilog_fun : i1, i1, i1
  ...
}

}
```

By and large, (1) this can lower to a given dialect's `extern` variant, and (2) if a user tries to lower dialect X definitions through conversion pass to dialect Y, it will fail unless dialect Y supports lowering dialect X as well.

Let me know if I'm off course here.

mortbopet, 2022-05-28:
For the sake of discussion - what do we actually gain by creating new Calyx primitive-defining operations?

That is, would a Calyx primitive defining operation be anything radically different than what we can already get from existing dialects + attaching attributes Calyx dialect attributes? As an example, reusing `hw.module`/`hw.module.extern` allows for leveraging the existing work on module parameterization - example IR:

```mlir
module {
  // Parameterizeable module with no body (extern). Staticness is communicated
  // through an additional dialect attribute.
  hw.module.extern<W: i32> @std_add(
    %a: !hw.int<#hw.param.decl.ref<"W">>, %b: !hw.int<#hw.param.decl.ref<"W">>)
    -> (out: !hw.int<#hw.param.decl.ref<"W">>) {
      // Calyx dialect attributes to reason about this module.
      calyx.static = 2
    }

  // CIRCT RTL dialect module with a body.
  hw.module @my_verilog_fun(%clk: i1, %reset: i1, %a: i32, %b: i32)
    -> (out: i32, done: i1) {
      // CIRCT RTL dialect stuff
    } {
      // Calyx dialect attributes to reason about this module.
      ...
    }

  calyx.program {
    calyx.component @C () -> () {
      %0:3 = calyx.instance @std_add<W = 32> : i32, i32, i32
      %1:5 = calyx.instance @my_verilog_fun : i1, i32, i32, i32, i1
      ...
    }
  }
}
```

rachitnigam, 2022-05-28:
I don't have a strong preference either way, especially if `hw.extern` can represent everything that `calyx.primitive` can. However, @mikeurbach noted in the original issue that we'd like to make `calyx.primitive` not be a black box in the future to enable co-optimization between dialects. If I misunderstood that this means, let me know. 

Either way, could be a good short term way to add primitives into the dialect even if we want to change it in the future

rachitnigam, 2022-05-28:
Also, if we do go with the `hw.extern` strategy, can someone outline/point me to documentation on how to do that?

mortbopet, 2022-05-28:
My hunch would be that a `calyx.primitive` would look a lot like the [calyx.instance in ODS](https://github.com/llvm/circt/blob/main/include/circt/Dialect/Calyx/CalyxStructure.td#L203), so that could be used as a baseline.

Next is the verifier of the `calyx.primitive` op. Again, will be fairly similar to the [`calyx.instance` verifier](https://github.com/llvm/circt/blob/main/lib/Dialect/Calyx/CalyxOps.cpp#L1221), with the change that the symbol should be defined by a `HWModuleLike` operation. Much of the verifying code could be shared between the two.

Finally, support is needed in `CalyxToHW`. I see that `calyx.instance` isn't included in `CalyxToHW` yet, so that cannot serve as a baseline. However, `calyx.primitive` would lower directly to a `hw.instance` operation, much like how arith binary/unary ops are lowered. i.e., compared to [`convertArithBinaryOp`](https://github.com/llvm/circt/blob/main/lib/Conversion/CalyxToHW/CalyxToHW.cpp#L337) the only difference would be that there is an arbitrary number of in- and output ports that need to be wired up.

rachitnigam, 2022-05-30:
Thanks for the outline @mortbopet! What do you think about the difference between `hw.extern` and a new `calyx.primitive` operator? Do you think it would be worth have a specific operator for Calyx or just rely on `hw.extern`?

rachitnigam, 2022-05-31:
I guess we'd want the outputs from `calyx.primitive_definition` to be the same format as `calyx.register` (a bunch of ports that can be used in assignments). I'd probably model `hw.primitive` in this way and see if that works with the rest of the `hw` dialect.

mikeurbach, 2022-06-01:
Just catching up on this convo, and I think I agree with everything said before, just one question:

> I think it's a good idea to create start off with creating a hw.primitive (instantiating) operation that targets HWModuleLike's.

> I'd probably model hw.primitive in this way and see if that works with the rest of the hw dialect.

These to mentions of `hw.primitive` threw me off, do those mean `calyx.primitive`? I see the discussion later moves back to the term `calyx.primitve`, e.g. "what the `calyx.primitive` operations does with the information...". To be clear, we are talking about one new primitive op in the Calyx dialect, which refers to HWModuleLike things?

closed 2022-08-04 by rachitnigam