llvm/circt #2935: [ExportVerilog] Port types (of lack thereof)

teqdruid, 2022-04-21:
`ExportVerilog` leaves off port types (e.g. `input [3:0] a` instead of `input wire [3:0] a`). This ends up producing a ton of warnings in questa (at least). IIRC, there was a specific reason for this behavior, right? Or was there not and we should just add `wire` where appropriate?

```
** Warning: foo.sv(4): (vopt-13314) Defaulting port 'x' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
```

I'm no SystemVerilog expert, but what can we do to avoid this particular warning (aside from changing the flags this project invokes questa with)? I'd imagine that this behavior is influenced by `default_nettype`? Do we need to explicitly set the default net type at the beginning of all our sv files?

darthscsi, 2022-06-07:
There is a flag to give per port types in the output.  I don't remember what it was called.  @uenoku ?

uenoku, 2022-06-17:
We have `disallowPortDeclSharing` option to force emission of port types to each line but it doesn't emit `wire`.

teqdruid, 2023-02-28:
I came up against this again today. I'm compiling with some code which rudely sets default_nettype so none and doesn't reset it, so CIRCT-generated code errors out.

@darthscsi / @seldridge Is there a particular reason we don't emit `wire` with each port? Would it be reasonable to introduce a lowering option to add this behavior? Would it be reasonable to add one which adds `default_nettype wire` to the beginning of every sv file?

darthscsi, 2023-02-28:
There should be an emission option.

teqdruid, 2023-02-28:
https://github.com/llvm/circt/pull/4737

closed 2023-02-28 by teqdruid