llvm/circt #2951: [StandardToHandshake] Support for partial lowering

Dinistro, 2022-04-22:
I’m working on a custom streaming dialect that encapsulates regions containing `arith` and `cf` operations. The most promising lowering target seems to be the handshake dialect, as this can be integrated into our existing infrastructure. 
Unfortunately, the streaming dialect cannot be lowered through std as this is not classical sequential code. While transforming the streaming operations directly to `handshake` is fine, I would like to reuse the transformations from `StandardToHandshake` for the encapsulated regions. 

Currently, the `StandardToHandshake` conversion pass only matches on `func::FuncOp`s and transforms them (and the content of their region) in more or less one big step to `handshake::FuncOp`s. There is no clear way of reusing only parts of this pass to transform code like the following:

```
module {
  handshake.func @opt(%cond: i1, %input: i32, %ctrl: none) -> (i32, none) {
    %true_val, %false_val = cond_br %cond, %input : i32
    sink %false_val : i32
    handshake.return %true_val, %ctrl : i32, none
  }
  
  handshake.func @filter(%ctrl: none) -> (i32, none) {
    %cond = arith.constant 0 : i1
    %val = arith.constant 42 : i32
    %res:2 = handshake.instance @opt(%cond, %val, %ctrl) : (i1, i32, none) -> (i32, none)
    handshake.return %res#0, %res#1 : i32, none
  }
}
```

Do you guys see any hope in refactoring or enhancing the `StandardToHandshake` pass to allow for such use cases?

mortbopet, 2022-04-22:
Cool! Do you mind sharing some of your streaming dialect IR? we've been discussing something like this for a while, so it's great to hear that progress may be on the horizon!

Just to make sure that i understand the example correctly, in the given case, you'd like to run the constant op transforms from `StandardToHandshake` to convert the `arith.constant`s to `handshake.constant`s that are connected to control, correct?

You're right in that `StandardToHandshake` is fairly monolithic. Some parts of it i see hard to reuse (such as adding branch/merge ops etc., since these are inherently tied to controlflow (`cf` dialect operations)). Other parts, such as [`connectConstantsToControl`]( https://github.com/llvm/circt/blob/main/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L180) could be used in a more general way.

One approach could be to make [`FuncOpLowering`](https://github.com/llvm/circt/blob/main/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L173) publicly available (and rename it, e.g. `HandshakeLowering`), and then modifying it's reliance on a [handshake::FuncOp](https://github.com/llvm/circt/blob/main/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L225) to instead be *any* operation which implements a region. I think this should be possible, since (AFAIR) that `handshake::FuncOp` is mostly used through iterating over its inner operations... which we might as well do in a more generic way :wink:.
Then you should ideally be able to pass it your operation which implements a region containing `arith,cf,...` code, and let it transform these.

CC @stephenneuendorffer

Dinistro, 2022-04-22:
> Cool! Do you mind sharing some of your streaming dialect IR? we've been discussing something like this for a while, so it's great to hear that progress may be on the horizon!

As I just started to work on this a week ago, the dialect is far from well designed. I might share some of the design after they converge to something useful. Furthermore, I need to check if I'm even allowed to publish these, as they are part of my Master's Thesis.

> Just to make sure that i understand the example correctly, in the given case, you'd like to run the constant op transforms from StandardToHandshake to convert the arith.constants to handshake.constants that are connected to control, correct?

Yes, this is exactly what I would like to do.

> One approach could be to make [FuncOpLowering](https://github.com/llvm/circt/blob/main/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L173) publicly available (and rename it, e.g. [...]
Then you should ideally be able to pass it your operation which implements a region containing arith,cf,... code, and let it transform these.

This sounds like a good plan. If this is something of general interest, I can try to implement these changes.

mortbopet, 2022-04-22:
> If this is something of general interest, I can try to implement these changes

If we can make the handshake lowerings more generic while providing utility to other lowerings, I see no reason not to!

mikeurbach, 2022-04-22:
+1 this is exciting to see. The StandardToHandshake lowering is indeed very monolithic, but like everything in the MLIR community, we only make changes when there is a real use-case driving them. Thanks for bringing a use-case to drive splitting up StandardToHandshake. I don't think there's anything fundamentally blocking it from being broken up and made useful to other dialects, it's just engineering effort to make that happen.

PR #2986 (2022-04-29): [StandardToHandshake] enable reuse by exposing region lowering
This change makes the lowering infrastructure used to lower `func::FuncOp` bodies available for other usages. As discussed in #2951 the lowering of the region is not dependent on the operation holding it. 

The newly added `lowerRegion` function expects a sub-type of `HandshakeLowering` which might have overridden member functions. The members are declared `virtual` to allow for such overriding without re-implementing the `lowerRegion` function. 

**Limitation**: Using this in a rewrite that changes the types of the entry block's ` [... cut]
closed 2022-04-29 by mortbopet