llvm/circt #2962: [ExportVerilog] Unnecessary temporary wires for single use values

uenoku, 2022-04-25:
input: 
```mlir
module attributes {circt.loweringOptions = "disallowLocalVariables"} {
  hw.module @Foo( %clock: i1, %opcode: i3) {
    %c-2_i3 = hw.constant -2 : i3
    %0 = comb.icmp eq %opcode, %c-2_i3 : i3
    sv.ifdef  "SYNTHESIS" {
    } else {
      sv.always posedge %clock {
        sv.if %0 {
          sv.error "Foo"
        }
      }
    }
    hw.output
  }
}
```
```verilog
module Foo(     // tmp2.mlir:2:3
  input       clock,
  input [2:0] opcode);

  `ifndef SYNTHESIS     // tmp2.mlir:5:5
  wire _GEN = opcode == 3'h6;   // tmp2.mlir:3:15, :4:10
    always @(posedge clock) begin       // tmp2.mlir:7:7
      if (_GEN) // tmp2.mlir:8:9
        $error("Foo");  // tmp2.mlir:9:11
    end // always @(posedge)
  `endif
endmodule
```

`_GEN` should be inlined to if condition since it is used only once.

uenoku, 2022-04-25:
`_GEN` is spilled because `%0` is used by `sv.if` in a different block. This logic originally comes from https://github.com/llvm/circt/commit/c29ca85d6d0890136558c0ae1ee4a1c039ede3be which intended to avoid creating wires in procedural regions. I'm wondering that restriction is still valid today.. I haven't come with an example which requires the restriction.

mikeurbach, 2022-04-25:
Would it be sufficient to use MLIR's notion of Dominance instead of the current check? For the narrow case of comb expressions and if-conditions, it seems like this would allow `_GEN` to be inlined. In general, I'm not sure the SV semantics here, and whether this is always legal.

uenoku, 2022-04-26:
> Would it be sufficient to use MLIR's notion of Dominance instead of the current check?

Yes, I think so. Considering that MLIR SSA values already ensure the dominance between use and def block, I believe we don't require the restriction. 

> In general, I'm not sure the SV semantics here, and whether this is always legal.

Yeah, I agree this concern.

mikeurbach, 2022-04-26:
> I believe we don't require the restriction.

So you mean to simply drop `if (user->getBlock() != opBlock)`? I hadn't thought about that. I was thinking more along the lines of changing that to `if (op.dominates(user))` (I forget the exact API, but it should be possible to set up a DominanceInfo for each HW module). Since we have graph regions, I don't think we are necessarily guaranteed that a definition will always dominate a use (by MLIR's definition of dominance in Dominance.h).

uenoku, 2022-04-26:
> if (user->getBlock() != opBlock)?
Yes, I was thinking about dropping that condition. 

> I hadn't thought about that. I was thinking more along the lines of changing that to if (op.dominates(user)) (I forget the exact API, but it should be possible to set up a DominanceInfo for each HW [...]

Yes, sorry for the my misword. I wanted to say that dominance between user and def *blocks* is guaranteed even in graph regions(this might be wrong). If so, since we are generating temporary wires at the beginning of blocks, we don't have to check that conditions.

mikeurbach, 2022-04-26:
> I wanted to say that dominance between user and def blocks is guaranteed even in graph regions(let me correct if wrong).

I see, I think this does make sense now that I think about it some more. The way we've carefully defined procedural regions and their nesting rules in ODS should make this check safe to remove.

dtzSiFive, 2022-05-04:
This seems fixed on latest main, FWIW, in particular with 959f3a4fbf8964fd0b4f7b7a7c850f291050bc9b from the linked PR.
Anything more here?

mikeurbach, 2022-05-05:
I think this can be close, @uenoku is there anything else here?

uenoku, 2022-05-05:
Thanks! It can be closed now.

closed 2022-05-05 by uenoku