llvm/circt #2985: Build rocket-chip or chipyard project

wangliwei-intel, 2022-04-27:
Is there an example/demo for showing how to build [https://github.com/chipsalliance/rocket-chip](url) or [https://github.com/chipsalliance/rocket-chip](url) with CIRCT/firtool?
Thanks.

wangliwei-intel, 2022-04-29:
@seldridge Thank you very much.
One more question, Does firtool take as input HiFIRRTL, LoFIRRTL or both?
-Liwei

seldridge, 2022-04-29:
It'll take whatever you throw at it. All the passes are there to handle anything from CHIRRTL on down. I.e., you can take whatever Chisel produces and compile it directly.  If you feed it Low FIRRTL, some passes may be no-ops.  Most common annotations/custom passes are also supported, but not everything.  (You can pass annotation files with repeated usage of `--annotation-file`.)

If you want an extremely experimental integration, there is also https://github.com/sifive/chisel-circt which gives you a `ChiselStage`-like API, but to compile using `firtool` under the hood.

wangliwei-intel, 2022-04-29:
@seldridge I tried your chisel-circt and chisel-circt-demo, they're nice.
Is't possible to compile rocket-chip with chisel-circt? of cause after replacing chisel3.xxx.xxxx with your circt.xxx.xxxx.
Thanks.
-Liwei

codelec, 2022-05-08:
It is possible to compile rocket-chip through firtool

wangliwei-intel, 2022-05-08:
@codelec Thank you.
Is there any publicly available example/repo  of rocket-chip built with firtool?

B. R.
Liwei

codelec, 2022-05-14:
https://github.com/codelec/chisel-circt/tree/nobb
```scala
object CIRCT extends App {
  val targetDir = new java.io.File("dgen")
  val argss: Array[String] = Array(
    "--target", "systemverilog",
    "--target-dir", targetDir.toString
  )
  val argsc: Array[String] = Array(
    "--chisel-output-file", "first.fir",
    "--no-run-firrtl",
//    "-T", "chipyard.TestHarness",
    "--target-dir", targetDir.toString
  )

  (new circt.stage.ChiselStage)
//  (new chisel3.stage.ChiselStage)
    .execute(argss, Seq(chisel3.stage.ChiselGeneratorAnnotation(() => new freechips.rocketchip.system.TestHarness()(new freechips.rocketchip.system.DefaultConfig ++ new freechips.rocketchip.subsystem.WithoutTLMonitors()))))
   // .execute(argsc, Seq(chisel3.stage.ChiselGeneratorAnnotation(() => new TestHarness()(new chipyard.SmallBoomConfig ++ new freechips.rocketchip.subsystem.WithoutTLMonitors()))))
    .map {
      case firrtl.annotations.DeletedAnnotation(_, a) => a
      case a => a
    }
}
```