llvm/circt #3023: Non-handhsake inter-module and host communication interfaces

JuanEsco063, 2022-05-03:
I could not find a specific issue dealing with this topic so I am just creating a new one. If there is, feel free to reference it here and I will close this one.

My questions is on the current status of adding a standardized interface wrapper to modules generated using the CIRCT flow. 

Currently, I am taking the RTL generated by CIRCT and adding all the additional components necessary to talk to the kernel via AXI manually. This allows me to connect AXI-based IP to it and more importantly, drive the kernel either via MMIO (requires sudo access) or even OpenCL using XRT. 

I know ESI from @teqdruid is looking to achieve something similar to AXI (interconnect modules) but I am not sure on the status of the project or if it will be AXI compatible in a way.

Thanks in advance!

JuanEsco063, 2022-05-05:
Thank you @mortbopet ! I will reach out to @makslevental !

Currently I have a hybrid approach. I start from Standard code and lower it to Verilog using CIRCT. This allows me to interface to BRAM easily by declaring the memories as ports (inout/output) in the Standard code (I manually linearize the code to only use 1D memories). I then created some AXI masters to perform data initialization using HLS. This is, loading data on the same BRAMs which will be used by the RTL generated by CIRCT. Then I use an AXI slave (also generated using HLS) to give the AXI masters the physical memory address offset and start a state machine that controls the start of the DMA via AXI master from host to the FPGA, start the execution of the RTL kernel generated by CIRCT, and then perform the final DMA from the FPGA back to host.

Feel free to ping me directly if you want further clarification!

makslevental, 2022-05-05:
@JuanEsco063 you and i should chat - how can i contact you? alternatively, my email is in my profile...

JuanEsco063, 2022-05-05:
@makslevental I will shoot you an email

teqdruid, 2022-05-09:
Sorry I haven't weighed in before now.

@mortbopet is correct -- this is exactly what ESI is designed for. Currently, ESI has support for building on-chip point-to-point interconnects (ready-valid wire protocol only) and a cosimulation interface for "off-simulated-chip". We're planning on building real off-chip stuff via ESI services, as @mortbopet has stated.

I think we've figured out a relatively simple way to implement ESI services, it's just a matter of doing the actual implementation. I have yet to break it out into individual issues though.

teqdruid, 2022-05-09:
That's more related to the stuff I'm currently working on in the MSFT dialect. (See my [LATTE'22 paper](https://capra.cs.cornell.edu/latte22/).) The first construct I'm planning on building is a broadcast operation, which will be able to interactively lower and lower with separate parameters on a per-(dynamic)-instance basis (rather than per-static-instantiation-site).

I'm in the planning stage for the broader project now, so I don't know when I'll get to it.

teqdruid, 2022-05-09:
The relationship between ESI and the sort of structures I'm building for the MSFT dialect are more and more blurry. The key distinctions as of now are that the MSFT ops aren't latency-insensitive and optimized for extreme physical design. Both ESI and the MSFT constructs are fundamentally communication constructs, though, so it may or may not make sense to merge them at some point.