llvm/circt #3028: [SV] Miss-complication by reordering side-effect ops

uenoku, 2022-05-03:
```
hw.module @Foo(%clock:i1, %a:i8, %b:i8) {
  %reg = sv.reg : !hw.inout<i8> 
  %fd = hw.constant 0x80000002 : i32
  sv.always posedge %clock {
    sv.bpassign %reg, %a : i8
    %x = sv.read_inout %reg : !hw.inout<i8>
    sv.bpassign %reg, %b : i8
    sv.fwrite %fd, "%d"(%x) : i8
  }
}
```

Current output:
```
module Foo(
  input       clock,
  input [7:0] a,
              b);

  reg [7:0] reg_0;

  always @(posedge clock) begin
    reg_0 = a;
    reg_0 = b;
    $fwrite(32'h80000002, "%d", reg_0);
  end // always @(posedge)
endmodule
```

This is incorrect because the value of `reg_0` should be equal to `a`

uenoku, 2022-05-03:
There are independent bugs at least in Prepare and ExportVerilog. 

1. Prepare
With disallowLocalVariable option, Prepare swaps `bpassign` and `read_inout` which is obviously incorrect. 
```
  sv.always posedge %clock {
    sv.bpassign %reg, %a : i8
    sv.bpassign %reg, %b : i8
    %0 = sv.read_inout %reg : !hw.inout<i8>
    sv.fwrite %c-2147483646_i32, "%d"(%0) : i8
  }
```
3. ExportVerilog
```
```

mikeurbach, 2022-05-03:
Is this introduced in https://github.com/llvm/circt/commit/300e24414e95a53bbf10ab10f50b836b7e31324a? That's the most recent change to read_inout ordering. I may have approved that PR to hastily.

+1 for introducing constructs in Prepare to Export can be simpler.

uenoku, 2022-05-04:
No, https://github.com/llvm/circt/commit/300e24414e95a53bbf10ab10f50b836b7e31324a is not source of this bug. I confirmed reordering happens before the commit. However, it seems https://github.com/llvm/circt/commit/300e24414e95a53bbf10ab10f50b836b7e31324a moves read_inout's position so I feel this might be doing something incorrect.

uenoku, 2022-05-04:
I tried this firtool on sifive/0/0/1~sifive/0/6/0. It has never been compiled correctly:pensive: