llvm/circt #3037: [Handshake] Support for tuple type

Dinistro, 2022-05-04:
Currently, a handshake function with multiple arguments introduces handshaking for each argument separately. In some cases this is not required, as the input arrives in the form of a tuple.

The main problem with supporting MLIR's builtin `TupleType` (https://mlir.llvm.org/docs/Dialects/Builtin/#tupletype) is that it does not come with standard operations to operate on such types (Rational link: https://mlir.llvm.org/docs/Dialects/Builtin/#tupletype). Therefore, we would have to introduce tuple access and packing operations. One potential use of these additions would look as follows (operation names to be defined):

```
handshake.func @types(%in: tuple<i64, i32, i64>, %arg1: none, ...) -> (tuple<i32, i64>, none) attributes {argNames = ["in0", "inCtrl"], resNames = ["out0", "outCtrl"]} {
  %a, %b, %c = handshake.unpack %in : (tuple<i64,i32,i64>) -> (i64,i32,i64)
  %sum = arith.addi %a, %c : i64

  %res = handshake.pack %b, %sum : (i32, i64) -> tuple<i32, i64>

  return %res, %arg1 : tuple<i32, i64>, none
}
```

Do you guys think that such additions to handshake are reasonable?

mortbopet, 2022-05-04:
Makes sense to me, and I think this would be nice to have for HLS flows such that we don't force users to unpack tuple elements at the top level (thus diverging from the function type of the source function).  

For implementing pack and unpack operations, a custom parser can be used to specify only the tuple type, given that the return type(s) can be inferred, i.e.:
```
%a, %b, %c = handshake.unpack %in : tuple<i64,i32,i64>
%res = handshake.pack %b, %sum : tuple<i32, i64> // or i32, i64
```

For lowering, I'd imagine this would lower almost identically to fork/join operations wrt. control signals.

Dinistro, 2022-05-04:
> Makes sense to me, and I think this would be nice to have for HLS flows such that we don't force users to unpack tuple elements at the top level (thus diverging from the function type of the source function).

This is exactly what I encounter with my thesis project. 

> For implementing pack and unpack operations, a custom parser can be used to specify only the tuple type, given that the return type(s) can be inferred, i.e.:

That's a good point. This should even remove the need for custom type-equality verifiers, IIRC.

mikeurbach, 2022-05-05:
So the inputs and outputs of a function might be a tuple, and you want to save on the handshaking logic by only having one handshake per tuple? And within the function, you might want to process chunks of the tuple independently? 

If I'm understanding I think it makes sense, but I think like any IR addition it needs a solid use-case. If the use-case is like the above, where you know certain IOs of a function always handshake together, this seems like a good way to represent that.

Have you thought about how this will lower? As Morten said, the fork/join logic should hopefully be reusable here. The types themselves can probably lower to a fixed width vector type. I was hoping we'd never see another addition to HandshakeToFIRRTL, but alas, the move off FIRRTL has never gotten momentum.

Dinistro, 2022-05-05:
> So the inputs and outputs of a function might be a tuple, and you want to save on the handshaking logic by only having one handshake per tuple? And within the function, you might want to process chunks of the tuple independently?
If I'm understanding I think it makes sense, but I think like any IR addition it needs a solid use-case. If the use-case is like the above, where you know certain IOs of a function always handshake together, this seems like a good way to represent that.

In my case, we connect an external stream to the handshaked circuit. This stream can produce one tuple per cycle which have a fixed size. So instead of unpacking this tuple externally and adding additional handshaking logic, I would like to use the tuples directly in a handshake circuit. 

> Have you thought about how this will lower? As Morten said, the fork/join logic should hopefully be reusable here. The types themselves can probably lower to a fixed width vector type. 

I think that `VectorType` would not work, as all the elements have the same type. The more fitting type to target seems to be `BundleType`, but that comes with names for each element. One might just use the original lowering strategy that introduces a `Bundle` for each handshaked input and wrap an additional bundle representing the tuple into it. 

> I was hoping we'd never see another addition to HandshakeToFIRRTL, but alas, the move off FIRRTL has never gotten momentum.

Yes, I would also prefer the direct flow to HW, but unfortunately I do not have the capacity to work on that.

Dinistro, 2022-05-05:
> I think that VectorType would not work, as all the elements have the same type. The more fitting type to target seems to be BundleType, but that comes with names for each element. One might just use [...]

One issue that I encountered with this approach is that bundles get lowered to separate inputs in Verilog. Not sure if this can be avoided though.

mikeurbach, 2022-05-05:
> adding additional handshaking logic

This is the key thing for me. We want to keep the IR design small and orthogonal, and making additions to simplify things for frontends needs to be balanced with complicating the IR and everything downstream that has to deal with the addition (HandshakeToFIRRTL, handshake-runner, etc.). But in this case, being able to pump an aggregate value into the handshake function, pull it apart, work on it, and pump out an aggregate value seems worth the cost. Moving an aggregate in/out with a single handshake is the killer feature IMO, not that it simplifies life for frontends. There is a reason nothing in-tree in upstream MLIR uses tuples.

> One issue that I encountered with this approach is that bundles get lowered to separate inputs in Verilog. Not sure if this can be avoided though.

This was the first thing I looked at when I joined the CIRCT project almost a year and a half ago! There has always been some interest in _not_ pulling apart aggregates at the FIRRTL level (this is the LowerTypes pass), but we've gotten this far without it. However, I think there is finally some momentum behind the "preserve aggregates" approach, which may be useful to you in the coming months. Until that happens, we should still be able to achieve the goal of one set of handshaking circuitry per tuple, even if they are flattened in the output verilog.

closed 2022-05-06 by Dinistro