llvm/circt #3111: lower-scf-to-calyx => valueGroupAssigns.end() && "No group evaluating value!"' failed

A-Dane, 2022-05-12:
When lowering lower-scf-to-calyx I'm getting:

circt-opt --allow-unregistered-dialect --lower-scf-to-calyx  -o=test_memory_3_calyx.mlir  test_memory_3_staticlogic.mlir

circt-opt: /mnt/scratch/tools/github/circt/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp:331: TGroupOp circt::ComponentLoweringState::getEvaluatingGroup(mlir::Value) [with TGroupOp = circt::calyx::GroupInterface]: Assertion `it != valueGroupAssigns.end() && "No group evaluating value!"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /mnt/scratch/tools/github/circt/build/bin/circt-opt --allow-unregistered-dialect --lower-scf-to-calyx -o=polygeist_circt_calyx/test_memory_3_calyx.mlir polygeist_circt_calyx/test_memory_3_staticlogic.mlir
 #0 0x00007fa78919b405 PrintStackTraceSignalHandler(void*) Signals.cpp:0:0
 #1 0x00007fa789198a84 SignalHandler(int) Signals.cpp:0:0
 #2 0x00007fa788a07520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #3 0x00007fa788a5ba7c __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #4 0x00007fa788a5ba7c __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #5 0x00007fa788a5ba7c pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #6 0x00007fa788a07476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #7 0x00007fa7889ed7f3 abort ./stdlib/./stdlib/abort.c:81:7
 #8 0x00007fa7889ed71b _nl_load_domain ./intl/./intl/loadmsgcat.c:1177:9
 #9 0x00007fa7889fee96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#10 0x00005652a5a0561f circt::BuildPipelineGroups::buildStageGroups(circt::staticlogic::PipelineWhileOp, circt::staticlogic::PipelineStageOp, mlir::Patte
[test_memory_3.zip](https://github.com/llvm/circt/files/8674539/test_memory_3.zip)
rnRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5d461f)
#11 0x00005652a5a05de5 circt::BuildPipelineGroups::PartiallyLowerFuncToComp(mlir::func::FuncOp, mlir::PatternRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5d4de5)
#12 0x00005652a59e845e circt::PartialLoweringPattern<mlir::func::FuncOp, mlir::OpRewritePattern>::matchAndRewrite(mlir::func::FuncOp, mlir::PatternRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5b745e)
#13 0x00007fa78949df6d mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<mlir::LogicalResult (mlir::Pattern const&)>) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRRewrite.so.15git+0x4ff6d)
#14 0x00007fa7895086a1 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>) (.constprop.0) GreedyPatternRewriteDriver.cpp:0:0
#15 0x00007fa78950a7b3 mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRTransformUtils.so.15git+0x607b3)
#16 0x00005652a59eb151 circt::SCFToCalyxPass::runOnOperation() (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5ba151)
#17 0x00007fa78942b19a mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2c19a)
#18 0x00007fa78942b8c0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2c8c0)
#19 0x00007fa78942cbe4 mlir::PassManager::run(mlir::Operation*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2dbe4)
#20 0x00007fa78a23215d performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>) (.constprop.0) MlirOptMain.cpp:0:0
#21 0x00007fa78a2324c1 processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, bool, bool, bool, bool, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, llvm::ThreadPool*) MlirOptMain.cpp:0:0
#22 0x00007fa78a232970 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x7970)
#23 0x00007fa78a234742 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x9742)
#24 0x00005652a547ca3c main (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4ba3c)
#25 0x00007fa7889eed90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#26 0x00007fa7889eee40 call_init ./csu/../csu/libc-start.c:128:20
#27 0x00007fa7889eee40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#28 0x00005652a547cd75 _start (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4bd75)

cgyurgyik, 2022-05-12:
Hi, thanks for pointing this out. Can you provide a minimal example please?

A-Dane, 2022-05-12:
Hi Chris,

 

Thanks, a zillion for doing this, and thanks for bearing with me – first time I send an issues to CIRCT.

 

Included is a reduced zip example the source is 

#define N 2

void test_memory_3(int a[N] ) {

  for (int i = 0; i < N; i++) { /* for loop with if creates the issue */

    if (a[0] == 42) a[0] = 0;

  }

}

 

The affine and staticlogic mlir are in the polygeist_circt_calyx dir. 

If there is a better / more preferred way to send test cases, please let me know – love to follow best practices.

 

Thanks,

Allan

cgyurgyik, 2022-05-12:
The best way is to provide a piece of IR from the language before the conversion pass. Since SCFToCalyx lowers both SCF and StaticLogic, then it may be one of those. I presume in this case it is StaticLogic. 

I suggest lowering your code to StaticLogic, and then try to remove as much code as possible while still triggering this error. You can then just copy and paste:

1. The minimal code example, 
2. What command you used, 
3. The resulting error.

After that, we can investigate exactly what is occurring. Also, for code snippets, please use 
\`\`\`
 // This is a code snippet
for (int i = 0; i < N; ++i) { foo(); }
\`\`\`

e.g.

```
// This is a code snippet
for (int i = 0; i < N; ++i) { foo(); }
```

Thanks :-).

A-Dane, 2022-05-13:
Thanks for the coaching!  I attempted to follow the guidelines, and looks like its StaticLogic creates issues.

``` 
int a[1];
// This is a code snippet
for (int i = 0; i < 2; i++) {  if (a[0] == 0) a[0] = 0;  }
``` 
when running 
/circt-opt --allow-unregistered-dialect --convert-affine-to-staticlogic -o=staticlogic.txt affine.txt

The following line in staticlogic.txt makes the scf-calyx error out.
```
// This is a code snippet in StaticLogic
memref.store %c0_i32, %0[%c0_1] : memref<1xi32>
````
Creates the resulting errors when running 
circt-opt --allow-unregistered-dialect --lower-scf-to-calyx -o=calyx.txt staticlogic.txt

circt-opt: /mnt/scratch/tools/github/circt/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp:331: TGroupOp circt::ComponentLoweringState::getEvaluatingGroup(mlir::Value) [with TGroupOp = circt::calyx::GroupInterface]: Assertion `it != valueGroupAssigns.end() && "No group evaluating value!"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /mnt/scratch/tools/github/circt/build/bin/circt-opt --allow-unregistered-dialect --lower-scf-to-calyx -o=calyx.txt staticlogic.txt
 #0 0x00007f591ff31dd5 PrintStackTraceSignalHandler(void*) Signals.cpp:0:0
 #1 0x00007f591ff2f454 SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f591f79e520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #3 0x00007f591f7f2a7c __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #4 0x00007f591f7f2a7c __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #5 0x00007f591f7f2a7c pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #6 0x00007f591f79e476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #7 0x00007f591f7847f3 abort ./stdlib/./stdlib/abort.c:81:7
 #8 0x00007f591f78471b _nl_load_domain ./intl/./intl/loadmsgcat.c:1177:9
 #9 0x00007f591f795e96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#10 0x0000563f171fe7e7 circt::BuildPipelineGroups::buildStageGroups(circt::staticlogic::PipelineWhileOp, circt::staticlogic::PipelineStageOp, mlir::PatternRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5ae7e7)
#11 0x0000563f171feef5 circt::BuildPipelineGroups::PartiallyLowerFuncToComp(mlir::func::FuncOp, mlir::PatternRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x5aeef5)
#12 0x0000563f171df27e circt::PartialLoweringPattern<mlir::func::FuncOp, mlir::OpRewritePattern>::matchAndRewrite(mlir::func::FuncOp, mlir::PatternRewriter&) const (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x58f27e)
#13 0x00007f592022a36d mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<mlir::LogicalResult (mlir::Pattern const&)>) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRRewrite.so.15git+0x4936d)
#14 0x00007f5920295631 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>) (.constprop.0) GreedyPatternRewriteDriver.cpp:0:0
#15 0x00007f5920297743 mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRTransformUtils.so.15git+0x60743)
#16 0x0000563f171e1937 circt::SCFToCalyxPass::runOnOperation() (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x591937)
#17 0x00007f59201be41a mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2c41a)
#18 0x00007f59201beab0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2cab0)
#19 0x00007f59201bfdd4 mlir::PassManager::run(mlir::Operation*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2ddd4)
#20 0x00007f5920f6203d performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>) (.constprop.0) MlirOptMain.cpp:0:0
#21 0x00007f5920f623a1 processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, bool, bool, bool, bool, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, llvm::ThreadPool*) MlirOptMain.cpp:0:0
#22 0x00007f5920f62850 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x7850)
#23 0x00007f5920f64622 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x9622)
#24 0x0000563f16c9bc4c main (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4bc4c)
#25 0x00007f591f785d90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#26 0x00007f591f785e40 call_init ./csu/../csu/libc-start.c:128:20
#27 0x00007f591f785e40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#28 0x0000563f16c9bf85 _start (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4bf85)

//code itself and outputs are below :
[test_memory_3.txt](https://github.com/llvm/circt/files/8683214/test_memory_3.txt)

[affine.txt](https://github.com/llvm/circt/files/8683291/affine.txt)

[staticlogic.txt](https://github.com/llvm/circt/files/8683293/staticlogic.txt)

Thanks!

cgyurgyik, 2022-05-13:
Thanks @A-Dane. Does anything immediately stand out to you @mikeurbach ? If not I'll look deeper into this, since I'm running into issues with this pass as well.

mikeurbach, 2022-05-13:
Thanks for posting this. I am away today and will take a deeper look early next week. Nothing immediately jumps out. For the record, this is very young code, and it's quite possible the example program is pushing into territory that code just doesn't handle. At the least, we should give a more informative error message rather than crashing the compiler.

cgyurgyik, 2022-05-14:
I'm working up a fix for this.

A-Dane, 2022-05-14:
Awesome! Thank you!

cgyurgyik, 2022-05-25:
Closing this. Feel free to open it.

closed 2022-05-25 by cgyurgyik