llvm/circt #3112: [StaticLogicToCalyx] basic block unused.

cgyurgyik, 2022-05-12:
```mlir
// test.mlir
func.func @func(%A: memref<4xi32>, %B: memref<4xi32>, %R: memref<4xi32>) {
  affine.for %i = 0 to 4 {
    %5 = affine.load %A[%i] : memref<4xi32>
    %6 = affine.load %B[%i] : memref<4xi32>
    %7 = arith.muli %5, %6 : i32
    affine.store %7, %R[%i] : memref<4xi32>
  }
  return
}
```

```
 ./build/bin/circt-opt -convert-affine-to-staticlogic test.mlir | ./bin/circt-opt --lower-scf-to-calyx -canonicalize
```

```mlir
"builtin.module"() ({
  "calyx.program"() ({
    "calyx.component"() ({
    ^bb0(%arg0: i32, %arg1: i1, %arg2: i32, %arg3: i1, %arg4: i32, %arg5: i1, %arg6: i1, %arg7: i1, %arg8: i1, %arg9: i32, %arg10: i2, %arg11: i1, %arg12: i32, %arg13: i2, %arg14: i1, %arg15: i32, %arg16: i2, %arg17: i1, %arg18: i1):
      %0 = "hw.constant"() {value = 1 : i32} : () -> i32
      %1 = "hw.constant"() {value = 4 : i32} : () -> i32
      %2 = "hw.constant"() {value = 0 : i32} : () -> i32
      %3 = "hw.constant"() {value = true} : () -> i1
      %4:2 = "calyx.std_slice"() {sym_name = "std_slice_2"} : () -> (i32, i2)
      %5:2 = "calyx.std_slice"() {sym_name = "std_slice_1"} : () -> (i32, i2)
      %6:2 = "calyx.std_slice"() {sym_name = "std_slice_0"} : () -> (i32, i2)
      %7:7 = "calyx.std_mult_pipe"() {sym_name = "std_mult_pipe_0"} : () -> (i1, i1, i1, i32, i32, i32, i1)
      %8:3 = "calyx.std_add"() {sym_name = "std_add_0"} : () -> (i32, i32, i32)
      %9:3 = "calyx.std_lt"() {sym_name = "std_lt_0"} : () -> (i32, i32, i1)
      %10:6 = "calyx.register"() {sym_name = "stage_1_register_0_reg"} : () -> (i32, i1, i1, i1, i32, i1)
      %11:6 = "calyx.register"() {sym_name = "stage_0_register_1_reg"} : () -> (i32, i1, i1, i1, i32, i1)
      %12:6 = "calyx.register"() {sym_name = "stage_0_register_0_reg"} : () -> (i32, i1, i1, i1, i32, i1)
      %13:6 = "calyx.register"() {sym_name = "while_0_arg0_reg"} : () -> (i32, i1, i1, i1, i32, i1)
      "calyx.wires"() ({
        "calyx.group"() ({
          "calyx.assign"(%13#0, %2) : (i32, i32) -> ()
          "calyx.assign"(%13#1, %3) : (i1, i1) -> ()
          "calyx.group_done"(%13#5) : (i1) -> ()
        }) {sym_name = "assign_while_0_init_0"} : () -> ()
        "calyx.comb_group"() ({
          "calyx.assign"(%9#0, %13#4) : (i32, i32) -> ()
          "calyx.assign"(%9#1, %1) : (i32, i32) -> ()
        }) {sym_name = "bb0_0"} : () -> ()
        "calyx.group"() ({
          "calyx.assign"(%4#0, %13#4) : (i32, i32) -> ()
          "calyx.assign"(%arg10, %4#1) : (i2, i2) -> ()
          "calyx.assign"(%12#0, %arg0) : (i32, i32) -> ()
          "calyx.assign"(%12#1, %3) : (i1, i1) -> ()
          "calyx.group_done"(%12#5) : (i1) -> ()
        }) {sym_name = "bb0_1"} : () -> ()
        "calyx.group"() ({
          "calyx.assign"(%5#0, %13#4) : (i32, i32) -> ()
          "calyx.assign"(%arg13, %5#1) : (i2, i2) -> ()
          "calyx.assign"(%11#0, %arg2) : (i32, i32) -> ()
          "calyx.assign"(%11#1, %3) : (i1, i1) -> ()
          "calyx.group_done"(%11#5) : (i1) -> ()
        }) {sym_name = "bb0_2"} : () -> ()
        "calyx.group"() ({
          "calyx.assign"(%8#0, %13#4) : (i32, i32) -> ()
          "calyx.assign"(%8#1, %0) : (i32, i32) -> ()
          "calyx.assign"(%13#0, %8#2) : (i32, i32) -> ()
          "calyx.assign"(%13#1, %3) : (i1, i1) -> ()
          "calyx.group_done"(%13#5) : (i1) -> ()
        }) {sym_name = "bb0_3"} : () -> ()
        "calyx.group"() ({
          "calyx.assign"(%7#3, %12#4) : (i32, i32) -> ()
          "calyx.assign"(%7#4, %11#4) : (i32, i32) -> ()
          "calyx.assign"(%10#0, %7#5) : (i32, i32) -> ()
          "calyx.assign"(%10#1, %7#6) : (i1, i1) -> ()
          "calyx.assign"(%7#2, %3) : (i1, i1) -> ()
          "calyx.group_done"(%10#5) : (i1) -> ()
        }) {sym_name = "bb0_4"} : () -> ()
        "calyx.group"() ({
          "calyx.assign"(%6#0, %13#4) : (i32, i32) -> ()
          "calyx.assign"(%arg16, %6#1) : (i2, i2) -> ()
          "calyx.assign"(%arg15, %10#4) : (i32, i32) -> ()
          "calyx.assign"(%arg17, %3) : (i1, i1) -> ()
          "calyx.group_done"(%arg5) : (i1) -> ()
        }) {sym_name = "bb0_5"} : () -> ()
      }) : () -> ()
      "calyx.control"() ({
        "calyx.seq"() ({
          "calyx.par"() ({
            "calyx.enable"() {groupName = @assign_while_0_init_0} : () -> ()
          }) : () -> ()
          "calyx.par"() ({
            "calyx.enable"() {groupName = @bb0_1} : () -> ()
            "calyx.enable"() {groupName = @bb0_2} : () -> ()
            "calyx.enable"() {groupName = @bb0_3} : () -> ()
          }) : () -> ()
          "calyx.par"() ({
            "calyx.enable"() {groupName = @bb0_1} : () -> ()
            "calyx.enable"() {groupName = @bb0_2} : () -> ()
            "calyx.enable"() {groupName = @bb0_3} : () -> ()
            "calyx.enable"() {groupName = @bb0_4} : () -> ()
          }) : () -> ()
          "calyx.while"(%9#2) ({
            "calyx.par"() ({
              "calyx.enable"() {groupName = @bb0_1} : () -> ()
              "calyx.enable"() {groupName = @bb0_2} : () -> ()
              "calyx.enable"() {groupName = @bb0_3} : () -> ()
              "calyx.enable"() {groupName = @bb0_4} : () -> ()
            }) : () -> ()
          }) {bound = 2 : i64, groupName = @bb0_0} : (i1) -> ()
          "calyx.par"() ({
            "calyx.enable"() {groupName = @bb0_4} : () -> ()
          }) : () -> ()
        }) : () -> ()
      }) : () -> ()
    }) {function_type = (i32, i1, i32, i1, i32, i1, i1, i1, i1, i32, i2, i1, i32, i2, i1, i32, i2, i1, i1) -> (), portAttributes = [{mem = {id = 0 : i32, tag = "read_data"}}, {mem = {id = 0 : i32, tag = "done"}}, {mem = {id = 1 : i32, tag = "read_data"}}, {mem = {id = 1 : i32, tag = "done"}}, {mem = {id = 2 : i32, tag = "read_data"}}, {mem = {id = 2 : i32, tag = "done"}}, {clk}, {reset}, {go}, {mem = {id = 0 : i32, tag = "write_data"}}, {mem = {addr_idx = 0 : i32, id = 0 : i32, tag = "addr"}}, {mem = {id = 0 : i32, tag = "write_en"}}, {mem = {id = 1 : i32, tag = "write_data"}}, {mem = {addr_idx = 0 : i32, id = 1 : i32, tag = "addr"}}, {mem = {id = 1 : i32, tag = "write_en"}}, {mem = {id = 2 : i32, tag = "write_data"}}, {mem = {addr_idx = 0 : i32, id = 2 : i32, tag = "addr"}}, {mem = {id = 2 : i32, tag = "write_en"}}, {done}], portDirections = -512 : i19, portNames = ["ext_mem0_read_data", "ext_mem0_done", "ext_mem1_read_data", "ext_mem1_done", "ext_mem2_read_data", "ext_mem2_done", "clk", "reset", "go", "ext_mem0_write_data", "ext_mem0_addr0", "ext_mem0_write_en", "ext_mem1_write_data", "ext_mem1_addr0", "ext_mem1_write_en", "ext_mem2_write_data", "ext_mem2_addr0", "ext_mem2_write_en", "done"], sym_name = "func", toplevel} : () -> ()
  }) {entryPointName = "func"} : () -> ()
}) : () -> ()
```

cgyurgyik, 2022-05-14:
```
   staticlogic.pipeline.stage start = 4 {
        memref.store %1, %arg2[%arg3] : memref<4xi32>
        staticlogic.pipeline.register 
      }
```

Since this just a store and doesn't register a result value, the pass completely ignores it.

cgyurgyik, 2022-05-14:
I'm working up a fix for this.

mikeurbach, 2022-05-16:
Thanks for the detailed reproducer here, will continue the discussion on the PR.

PR #3130 (2022-05-19): [SCFToCalyx] Add special case for pipeline stages with side effects only.
This is a hot fix for memref::StoreOps to be used unconditionally in a pipeline stage. I'm not really convinced this is the best approach, however I'm also hesitant to commit to any large-scale changes before we separate {SCF, StaticLogic}ToCalyx conversions.

Closes #3112, #3111.
closed 2022-05-19 by cgyurgyik