llvm/circt #3121: Discrepancy between behavioral simulation and post-implementation simulation/ hardware execution

JuanEsco063, 2022-05-12:
Hello,

I am using the most current version of CIRCT and I am having issues getting even some "basic" RTL to work properly. The behavioral simulation in Vivado 2021.2 is working correctly, but the post-implementation timing simulation shows incorrect behavior. I even tried the sample dot.mlir code from the circt/integration_test/Dialect/Handshake/ folder and using the driver code provided I get the same results. Below are screenshots of simulating the RTL generated from dot.mlir using the provided lowering commands:

Behavioral simulation:
[image]
[image]

Post-implementation timing simulation:
[image]

When I programmed the FPGA, the hardware was effectively not working. In both cases, it seems the RTL halts executing very early on.

Doing some debugging, I noticed in both the example code and my design, the multiplication module (arith_multi) is combinational. I do not know if this is the expected behavior but tracing things, the output of the multiplier is never computed in post-implementation simulation or in actual hardware (checked using an ILA).  To make matters worse, the operation is mapped to a DSP with the clock tied to 0 so it never completes any computation. 

Below are screenshots of the multi modules in both RTL and Synthesis schematic:

RTL:
[image]

Synthesized
[image]

Any insights if I am doing anything wrong would be appreciated.

mikeurbach, 2022-05-16:
Hi @JuanEsco063, to be honest, it sounds like you have gotten further along than anyone else (i.e. testing post-synthesis FPGA results with this flow). 

> Doing some debugging, I noticed in both the example code and my design, the multiplication module (arith_multi) is combinational. I do not know if this is the expected behavior but tracing things, [...]

It is intentional for the `arith.mul` operation to lower to a `comb.mul` operation, and eventually emit the System Verilog `out = left * right`. Whether or not that is desirable is an open question. So far, there hasn't been really any effort in the CIRCT project to control how the emitted System Verilog is synthesized, so let me put it this way: what would be desired here?

It sounds like the RTL code was mapped to a DSP by Vivado, but apparently it was mapped in a broken way. Is there some coding style we could add to the System Verilog emitter for Xilinx based FPGAs that would emit multiplications Vivado would handle correctly? Is this a place where the separate thread about emitting external primitives with CIRCT could help? Since this is untrodden territory, it seems like we'd need to design something from first principles in order to solve this problem the right way.

JuanEsco063, 2022-05-17:
@mikeurbach Thank you for your comment! Going over your questions:

> so let me put it this way: what would be desired here?

On a high level view, maybe 2 options would be useful: be able to force either synchronous or combinational logic as needed,  or be able to set arbitrary latency for the operation (either combinational or synchronous). Similar to how you can modify the files at the FIRRTL stage to change the R/W latency of memory modules. Isolating the arith_mult module and running tests on it in as a standalone project, it seems that with enough delay between input changes, even post-implementation simulation works correctly (see below):

Behav:
[image]
Post-synth
[image]
Post-impl
[image]

As you can see, after some transitioning time, the output of the multiplier does converge to the right value, even with purely combinational logic.

> Is there some coding style we could add to the System Verilog emitter for Xilinx based FPGAs that would emit multiplications Vivado would handle correctly? Is this a place where the separate thread [...]

I am not sure about this part since it seems the problem goes beyond just the multiplication module. The above observation led me to do further tracing on the dot example and somewhere along the way during synthesis, some logic is being optimized away for the address generation logic for the load operation.

The RTL analysis schematic of the generated SystemVerilog for dot shows a data-path for the input in0 of the multiplier that roughly goes something like: load0 (addr) -> buf 126 (addr) -> mem1 (data) -> buf 6 (data) -> load 0 (data) -> buf 127  (data) -> mult0. See image below for more details:
[image]

This leads to a correct execution of the behavioral simulation (see below):

[image]

However, after synthesis, the module "load0" seems to be completely missing and the path for the input in0 of the multiplier is closer to mem1 (data)-> buf 6 (data) -> buf 127 (data)  -> mult0. (See below, I have included a few extra buffers where the address generation is supposed to be taking place)

[image]

The post-synthesis timing simulation shows that no address is ever generated, not even in the initialization stage where the RTL is supposed to be storing constants to the memories. Below is the waveform of the post-synthesis timing simulation. All handshake signals have been removed by the synthesis tool for the mult module, and I have manually removed all the valid/ready signals of the other modules (buf, and mem) for clarity since they were always a constant value (wither 0 or 1).

[image]

At this point I am unsure at what level a solution can be implemented. 

I am testing the matmul example in the same example folder since we worked on it before and at some point it was working correctly.

JuanEsco063, 2022-05-17:
For some reason, it seems the RTL generated for matmul.mlir is also broken for post-synthesis and post-implementation simulations (and HW implementations)
Behavioral (works as expected)
[image]
Post-synthesis (does not work)
[image]

@mikeurbach has anyone else tried simulating the post-synthesis RTL recently? Maybe using Altera FPGA's/software. I might just be doing something wrong.

mikeurbach, 2022-05-17:
Thanks as always for providing detailed information.

> has anyone else tried simulating the post-synthesis RTL recently?

Not that I am aware of, either recently or potentially ever.

> I am testing the matmul example in the same example folder since we worked on it before and at some point it was working correctly.

Just to clarify, this worked correctly post-synthesis/post-implementation?

> On a high level view, maybe 2 options would be useful: be able to force either synchronous or combinational logic as needed, or be able to set arbitrary latency for the operation (either [...]

Thanks for trying to come up with a proposal here. I think I understand both options, but let me try to summarize to be sure:

Forcing synchronous or combinational logic would require two lowering styles, where currently there is one (all arithmetic ends up in the comb dialect). We could have sequential multipliers (potentially built out of comb.mul and seq.compreg), which could be an alternative. The handshaking circuitry would have to be sensitive to this, and wait for the sequential multiply to finish before outputs becoming valid and inputs becoming ready.

Setting arbitrary latency, like the R/W attributes on FIRRTL memories, would require a mechanism to specify how many cycles a handshake component would execute before outputs can become valid and inputs can become ready.

Both of these cases seem fairly unappealing, because we are making handshake components latency sensitive, while one of the big design goals of handshake is latency-insensitivity.

That said, I think we all want to see handshake be useful in more than just behavioral simulations, so we will need to find a solution here. My gut feeling is now that we are crossing from just behavioral simulation into post-synthesis/post-implementation simulations (and eventually programming real FPGAs), we need to move to a more sophisticated lowering path than just dumping out `comb.mul` for multiply. Modeling and targetting DSPs in a first class way, for example, could be a way forward.

JuanEsco063, 2022-05-18:
> Just to clarify, this worked correctly post-synthesis/post-implementation?
Yes, the matmul example used to work correctly in all stages of simulation and implemented in previous iterations of CIRCT. It was a long time ago, probably before the transition into dialects like comb.

> Both of these cases seem fairly unappealing, because we are making handshake components latency sensitive, while one of the big design goals of handshake is latency-insensitivity.
I see your point and agree it would take back from the original appeal of the handshake interface. How does this reconcile with dialects such as FSM? Read/Valid signals would be asserted only when the state machine is at the initial/final state and would depend on the number of states/path taken. Could something like this be used to model a multiplier that takes several clocks to compute the result?